klonyyy/MCUViewer #5: STMViewer does not update variables

aareyes96, 2023-07-09:
I'm using Windows.opens a console window that closes immediately and does not update the variable types.

[image]

[image]

klonyyy, 2023-07-09:
Hello, 
I'm guessing you don't have GDB installed. You can get it with MinGW. Please let me know if it solved the problem. Also please remember to add it to your path if it is not added automatically on install.

aareyes96, 2023-07-12:
Hello @klonyyy . Thanks for your reply. I have GDB installed with mingw64.
[image]
[image]

I just uncompress this file. and copy to C:/. 
[image]
Also I added to PATH enviroment variable

Thanks in advance

klonyyy, 2023-07-12:
Hello,
Could you attach the log file after a unsuccessful update attempt? It should be located under:
C:\Users\<username>\AppData\Roaming\STMViewer\logs

Could you also attach your *.elf file?

aareyes96, 2023-07-12:
Hello @klonyyy . The file is empty.
[image]

aareyes96, 2023-07-12:
[test_again.zip](https://github.com/klonyyy/STMViewer/files/12028606/test_again.zip)

Tharazi97, 2023-07-12:
Hi, I am facing the same issue. I also tried with release 0.1.1 with the same result.

klonyyy, 2023-07-12:
> Hello @klonyyy . The file is empty. [image]

Did you try it after trying to update the variables? 

Can you run the gdb using the console? Could you make sure it works by typing "gdb" in the console?

klonyyy, 2023-07-12:
It seems there is a problem with GDB versions before 12.1. I'm using 12.1 but later 13.2 should work as well. Could you try installing the newer version?

aareyes96, 2023-07-12:
[image]
[image]

@klonyyy Great !!!! Now is working !!!. I am very excited to use your software.. I'm using STLink-V2 clone. 

I attach the link for download the MingW with GDB 13.2 

https://github.com/brechtsanders/winlibs_mingw/releases/download/13.1.0-16.0.5-11.0.0-ucrt-r5/winlibs-x86_64-posix-seh-gcc-13.1.0-llvm-16.0.5-mingw-w64ucrt-11.0.0-r5.zip

Very Thanks !!!

klonyyy, 2023-07-12:
Glad it worked, thanks! I will add a proper note in the readme in the next update.

closed 2023-07-12 by klonyyy