klonyyy/MCUViewer #6: Hello

Candas1, 2023-07-11:
Hi,

I came across your project, I am really interested in using it as I am also very frustrated with STM Studio.
Seing [this](https://github.com/devanlai/webstlink) project a few months ago, I thought it would be great to have a web version of what you are doing, but now I see it's a lot of work.
Will your project work with any version of the Stlink ? 

I would like to promote your project [here](https://community.simplefoc.com/t/using-stm-studio-for-variable-monitoring/3177/8) if you don't mind.

klonyyy, 2023-07-11:
Hello! 
To be honest I don't have any experience in web development, so at this moment I don't think it's possible. 

I tested STMViewer with ST-Link V2 (the one from nucelo development boards, V3-MINI and V3-MINIE. You can refer to the stlink library that I'm using (it's listed in the end of the readme file) for more information on supported programmers. 

Regarding the promotion - I don't mind at all. Thanks!

klonyyy, 2023-07-11:
Let me know if you encounter any problems. I'm closing the issue for now.

closed 2023-07-11 by klonyyy

Candas1, 2023-07-26:
Hi,

Just to confirm, it works well for me with the chinese clones of Stlink V2.1, I even used it with a GD32 MCU.