klonyyy/MCUViewer #7: Unobserve variables from ELF

dskibarko, 2023-07-26:
Hello!
The problem observe with attached ELF-file & STMViewer 0.1.2.
Before subscribe this ticket ELF-file was seccesfully tested using J-Scope software.
Please investigate observed problem.

[WCH1.zip](https://github.com/klonyyy/STMViewer/files/12172910/WCH1.zip)

dskibarko, 2023-07-27:
Hello,

Seems structure of data do not parced?

I use Windows version tool and try scope UINT16 values

Switches_Raw.SH_FL_OVL_adc
Switches_Raw.SH_FL_BRK_adc

defined in this this C-language structure
typedef struct
{                           
    UINT16 U_SH_SHORTCUT_adc;
    UINT16 SH_FL_BRK_adc;     
    UINT16 SH_FL_OVL_adc;     
    UINT16 SH_FR_BRK_adc;     
    UINT16 SH_FR_OVL_adc;     
    UINT16 SH_RL_BRK_adc;     
    UINT16 SH_RL_OVL_adc;     
    UINT16 SH_RR_BRK_adc;   
    UINT16 SH_RR_OVL_adc;  
} SWITCHES_ADC_T;

klonyyy, 2023-07-27:
Just tried it and it works:
[image]
Probably its an issue related to #5. Please let me know if updating GDB helped.

dskibarko, 2023-07-27:
Ups.... 
So, about one week late i can try to reproduce more accurate and then inform you.

dskibarko, 2023-08-09:
Hello!
Not work!
[image]

Scenario for reproduce
1. Select ELF-file located in 
c:\hgLOCAL\BDFKE2023_Firmware\OUTPUT_WCH32\Objects\BDFKE2023_WCH.elf 
[BDFKE2023_WCH.zip](https://github.com/klonyyy/STMViewer/files/12298772/BDFKE2023_WCH.zip)
2. Press "ADD VARIABLE" --> variable with default name new0 appear in the list
3. Rename new0 to KWPcs and then press "Update variable address". --> the result is NOT FOUND

Have you any idea for additional test on my side?

klonyyy, 2023-08-09:
Hello, have you updated the GDB to version 12.1 or higher?

dskibarko, 2023-08-09:
Was upgraded to 12.1 and work fine now! Thanks!

closed 2023-08-09 by klonyyy