klonyyy/MCUViewer #9: Problem with changing variables name

indrekoidekivi, 2023-08-22:
Hi!
Small issue - after adding a new variable I am unable to rename it using GUI. I am able to write a new name after double click, but I havent found the way to save it - enter does not work and if I click away then the variable changes back to default "new0".

As a workaround I can create new variable, save config, rename variable in .cfg file and then open config.

I am on Ubuntu 20.04.6 LTS.

Overall its a great tool, thank you!

klonyyy, 2023-08-22:
Hello!

Could you record a short gif, or a video of the behavior? I haven't stumbled upon something like that yet. Also could you try the devel branch?

indrekoidekivi, 2023-08-22:
[image]
So here I add variable, change name and hit enter. Var name changes back to new0 after clicking. I havent had chance to try devel branch yet.

klonyyy, 2023-08-22:
Thanks for the recording. Could you try the attached devel installer to see if the problem persists? 
[STMViewer_installer_devel.zip](https://github.com/klonyyy/STMViewer/files/12410434/STMViewer_installer_devel.zip)

indrekoidekivi, 2023-08-24:
Yes, same issue with devel installer. On windows it works well.

indrekoidekivi, 2023-09-01:
Yes sure, sorry for late reply!
[stmviewer_ldd.txt](https://github.com/klonyyy/STMViewer/files/12496986/stmviewer_ldd.txt)

klonyyy, 2023-09-04:
It seems my output is exactly the same except some libraries are loaded from "/usr/lib/" in your case whereas all of my libraries are loaded from /lib/. Not sure if it could affect anything though. 

One more question - if you add more variables is the default (randomly generated) variable color always white? 

Could you also mention the installation process?

indrekoidekivi, 2023-09-08:
Interestingly yesterday evening the variable name changing randomly happened to work while I had project opened, I didnt have time to investigate at the time though, today morning its back to not working...hmm

The colors are every time random, so no not always white.

Installation process was exactly like described in the readme, `sudo apt install ./STMViewer-x.y.z-Linux.deb` etc.

klonyyy, 2023-09-12:
Thanks for the additional info! Let me know if you're able to make it work again (and under what circumstances), in the meantime I'll check if there's something that could cause such behavior.

klonyyy, 2024-02-10:
Closing due to inactivity. Please reopen if the problem persists.

closed 2024-02-10 by klonyyy