klonyyy/MCUViewer #17: Compiling STMViewer on ArchLinux

wimalopaan, 2023-10-23:
Just tried to compile/install STMViewer on arch linux with no luck from master.

I get the followin errors:

```
In Datei, eingebunden von /home/lmeier/Projekte/STMViewer/STMViewer/src/TargetMemoryHandler/StlinkHandler.hpp:4,
von /home/lmeier/Projekte/STMViewer/STMViewer/src/TargetMemoryHandler/StlinkHandler.cpp:1:
/home/lmeier/Projekte/STMViewer/STMViewer/src/TargetMemoryHandler/ITargetMemoryHandler.hpp:14:33: Fehler: »uint32_t« wurde nicht deklariert
14 |         virtual bool readMemory(uint32_t address, uint32_t* value) = 0;
|                                 ^~~~~~~~
/home/lmeier/Projekte/STMViewer/STMViewer/src/TargetMemoryHandler/ITargetMemoryHandler.hpp:14:51: Fehler: »uint32_t« wurde nicht deklariert
14 |         virtual bool readMemory(uint32_t address, uint32_t* value) = 0;
|                                                   ^~~~~~~~
/home/lmeier/Projekte/STMViewer/STMViewer/src/TargetMemoryHandler/ITargetMemoryHandler.hpp:15:34: Fehler: »uint32_t« wurde nicht deklariert
15 |         virtual bool writeMemory(uint32_t address, uint8_t* buf, uint32_t len) = 0;
|                                  ^~~~~~~~
/home/lmeier/Projekte/STMViewer/STMViewer/src/TargetMemoryHandler/ITargetMemoryHandler.hpp:15:52: Fehler: »uint8_t« wurde nicht deklariert
15 |         virtual bool writeMemory(uint32_t address, uint8_t* buf, uint32_t len) = 0;
|                                                    ^~~~~~~
/home/lmeier/Projekte/STMViewer/STMViewer/src/TargetMemoryHandler/ITargetMemoryHandler.hpp:15:66: Fehler: »uint32_t« wurde nicht deklariert
15 |         virtual bool writeMemory(uint32_t address, uint8_t* buf, uint32_t len) = 0;
|                                                                  ^~~~~~~~
/home/lmeier/Projekte/STMViewer/STMViewer/src/TargetMemoryHandler/StlinkHandler.hpp:15:14: Fehler: »bool StlinkHandler::readMemory(uint32_t, uint32_t*)« als »override« markiert, überschreibt aber nicht
15 |         bool readMemory(uint32_t address, uint32_t* value) override;
|              ^~~~~~~~~~
/home/lmeier/Projekte/STMViewer/STMViewer/src/TargetMemoryHandler/StlinkHandler.hpp:16:14: Fehler: »bool StlinkHandler::writeMemory(uint32_t, uint8_t*, uint32_t)« als »override« markiert, überschreibt aber nicht
16 |         bool writeMemory(uint32_t address, uint8_t* buf, uint32_t len) override;
```

klonyyy, 2023-10-23:
It seems the `#include <cstdint>` is missing. Could you add it to the ITargetMemoryHandler.hpp and let me know if the compilation succeded?

wimalopaan, 2023-10-23:
Had to add the include also to `Variable.hpp`. Now compiles fine.

However, starting vields an unresolved libspdlog.so.1.11. On my system spdlog-1.12 is installed. Maybe the version is hardcoded somewhere?

wimalopaan, 2023-10-23:
After starting (and stopping) I get the following messages on the shell:

```
[2023-10-23 11:04:18.223] [logger] [info] Starting STMViewer!
[2023-10-23 11:04:18.223] [logger] [info] Version: 0.2.0
[2023-10-23 11:04:18.223] [logger] [info] Commit hash 4ee85552d1f51cd0c9ba6f5eca58f9c9a4d09d59
./chips: No such file or directory
./chips: No such file or directory
[2023-10-23 11:04:26.166] [logger] [info] Exiting trace plot handler thread
[2023-10-23 11:04:26.209] [logger] [info] Exiting GUI main thread
[2023-10-23 11:04:26.227] [logger] [info] Closing STMViewer!
Speicherzugriffsfehler (Speicherabzug geschrieben)

```
Starting it from my home-dir gives me a weird gui. Starting it from the source-dir the gui looks ok.

klonyyy, 2023-10-23:
Thanks for the info. It seems the ./chips directory is not present in the executable directory. Could you copy it manually and see if the error goes away? It's located in third_party/stlink/ folder. 

The weird gui is due to the fact imgui.ini file should be placed in the folder the executable is run from. If its not detected by the library its going to default to some random window sizes and positions.

klonyyy, 2023-10-23:
Could you also let me know which Arch Linux you are currently using? I will make a VM an try to reproduce and fix these problems

wimalopaan, 2023-10-23:
> Could you also let me know which Arch Linux you are currently using? I will make a VM an try to reproduce and fix these problems

Arch Linux is a rolling distribution, so there is no version information.

wimalopaan, 2023-10-23:
> Thanks for the info. It seems the ./chips directory is not present in the executable directory. Could you copy it manually and see if the error goes away? It's located in third_party/stlink/ folder.

Hm, the error remains. 

```
lmeier  ~  ls -ld .chips
drwxr-xr-x 2 lmeier users 4096 23. Okt 13:27 .chips
lmeier  ~  /usr/local/STMViewer/STMViewer
[2023-10-23 13:29:24.420] [logger] [info] Starting STMViewer!
[2023-10-23 13:29:24.420] [logger] [info] Version: 0.2.0
[2023-10-23 13:29:24.420] [logger] [info] Commit hash 4ee85552d1f51cd0c9ba6f5eca58f9c9a4d09d59
./chips: No such file or directory
./chips: No such file or directory
[2023-10-23 13:29:32.398] [logger] [info] Exiting trace plot handler thread
[2023-10-23 13:29:32.414] [logger] [info] Exiting GUI main thread
[2023-10-23 13:29:32.424] [logger] [info] Closing STMViewer!
Speicherzugriffsfehler (Speicherabzug geschrieben)
lmeier  ~                                                

```

> 
> The weird gui is due to the fact imgui.ini file should be placed in the folder the executable is run from. If its not detected by the library its going to default to some random window sizes and positions.

Yes, copying manually fixed the weird gui layout.

klonyyy, 2023-10-23:
Sorry I accidentally added the './' prefix and I think this is why you got it wrong. The directory is called 'chips' not '.chips'

wimalopaan, 2023-10-23:
> Sorry I accidentally added the './' prefix and I think this is why you got it wrong. The directory is called 'chips' not '.chips'

Oh, my fault!
Perfekt now, but SEGV at the end remains.

klonyyy, 2023-10-23:
I think it might be related to the spdlog, as after the last message the spdlog flush() is called. I will look into that when I have a working VM with Arch. Thanks for the feedback and feel free to let me know in case any other issues arise!

closed 2023-12-19 by klonyyy