klonyyy/MCUViewer #18: Add drop-down-list to select a variable to watch

wimalopaan, 2023-10-23:
One has to manually enter the full name (mangled) of a variable to watch. If one use C++ with templates this is tedious, because one has to look into the .elf-file to get the mangled names: e.g. `_ZN4GFSMI7DevicesI10ESC_FOC_016ConfigN3Mcu3Stm9Stm32G431EEE10mDebugTickE` is such a name in my case.

It would be beautiful to get a drop-down-list with the de-mangled names: e.g.  `GFSM<Devices<ESC_FOC_01, Config, Mcu::Stm::Stm32G431> >::mDebugTick`

klonyyy, 2023-10-24:
It should be possible to just use the object instance name, without specyfing template parameters - have you tried that? A simplified example that I could try would be helpful as well.

wimalopaan, 2023-10-24:
In the above example `GFSM<Devices<ESC_FOC_01, Config, Mcu::Stm::Stm32G431> >` is the instantiated template and `mDebug` is a static member inside of that template-class. Do you mean, the name `mDebug` will suffice?

klonyyy, 2023-10-24:
I mean if you have an object instance for example `GFSM<Devices<ESC_FOC_01, Config, Mcu::Stm::Stm32G431> > test; ` you should be able to use `test.mDebug`.

wimalopaan, 2023-10-24:
> I mean if you have an object instance for example `GFSM<Devices<ESC_FOC_01, Config, Mcu::Stm::Stm32G431> > test; ` you should be able to use `test.mDebug`.

Ok, now I understand.
No, there is no instance of that template-class: all its members are static.

Would it be passible to get a drop-down-list of all objects with static storage for the menu from the `.elf`-file using de-mangled names? This would be very helpful. If you need an example, I will provide one.

klonyyy, 2023-10-24:
I think I was able to reproduce it, thanks. For now it its not possible, however I will look for an easy solution if there is one. If you want to quickly show some data, you can consider copying it's value to a global variable. I'm aware its ugly but maybe this is something that will make your work faster.

wimalopaan, 2023-10-24:
Well, that would be possible, but I think, I stick to the way copying the name from the `.elf` file.

Maybe another way would be to generate the STMViewer-config-file via some script parsing the `.elf`.

closed 2024-02-11 by klonyyy