klonyyy/MCUViewer #20: High speed plotting

Vasili-Sk, 2023-11-14:
Can we use STM logger (STtrace) library for buffered high data rate plotting? i'm also in motor control and sine is barely visible there :)

klonyyy, 2023-11-14:
Hello, could you elaborate on the topic, or show the problem you're facing? I'm not sure I understand the question.

Vasili-Sk, 2023-11-14:
STM Studio had special libraries for logging data from inside chip, this way you can synchronise it to end of FOC routine, for example and have clearer data at PWM speed
[image]
These are STM libraries. 
[image]
when you click this little square in stm studio it changes logger mode to buffered, also in settings you can setup how much calls to DumpTrace to be skipped. DumpTrace is a function you need to call from your code that saves data.

klonyyy, 2023-11-15:
I must admit I wasn't aware of this feature. STMStudio still keeps surprising me. I'd have to research this topic, but right now If you'd like a synchronous way of logging data you should try the Trace Viewer and SWO output. I'll keep in mind to try it out and see how it works/how it is implemented in STMStudio. Thanks!

Vasili-Sk, 2023-11-15:
yeah i wish but my boards dont have  SWO anymore since i found this thing and always  used it :P

klonyyy, 2023-11-15:
Got it. In this case I don't think it's possible to synchronize the readings currently. It seems like a very interesting feature, but most probably it will take some time for me research and implement it. Feel free to contribute on this feature if you have some free time, though! :)

dzid26, 2024-05-03:
There might be lower hanging fruits:
1. Different acquisition sample times per variable would be great 
2. Or a way to quickly disable some variables with e.g. checkboxes or ctrl+select - on the variable list - without the need to modify plots definition - to boost sampling time of other variables. 

Also, it would be good to know print real sampling time and make it red if it is lower than desired. :)

cickok, 2025-02-06:
Hello there,

this feature is so called "snapshot mode" and it is really useful for exaple PID controller tuning. We (automatization and motor contorl engineers) will be really appreciated if this feature will be implemented :)

BTW this tool is very nice, much more than original STM studio (which is moreover not supported by new elf v5 or something like that).

Good job and best regards.

klonyyy, 2025-02-06:
I'll give more info on this topic soon, for now I can say it will be integrated in MCUViewer ;)

klonyyy, 2025-05-22:
The MCUViewer now supports the recorder module [MCUViewer 1.2.0](mcuviewer.com/#downloads)

closed 2025-05-22 by klonyyy