klonyyy/MCUViewer #21: Any reason not to run on macos?

VIPQualityPost, 2023-11-15:
Trying to build gives:
STMViewer/src/main.cpp:33:2: error: "Your system is not supported!"

I can build on linux but my daily driver is macos, I do a lot of development on there. 
If there is no good reason, I will spend some time to see if I can make it work.

klonyyy, 2023-11-15:
Hello! To my best knowledge it should be possible to build STMViewer on macos, with a few changes in the CMakelists.txt and the code itself. I haven't done it yet as I'm not really familiar with macos, but if you're willing to contribute on this topic it'd be greatly appreciated!

margce, 2023-11-17:
Another daily macos user here, I can try and build too and report my findings. I'm running on a M1 machine just for reference.

margce, 2023-11-17:
I'm not familiar with compiling apps for desktop but i can still give it a go. I tried adjusting the CMakelists.txt and I was able to compile but not link. Like I said I'm running macos on a M1 macbook pro so I've installed stlink-org utils using homebrew which install v1.7.0 but for some reason some features are missing.

When trying to link it would come up with the following error
`ndefined symbols for architecture arm64:
  "_init_chipids", referenced from:
      StlinkHandler::StlinkHandler() in StlinkHandler.cpp.o
      StlinkHandler::StlinkHandler() in StlinkHandler.cpp.o
      StlinkTraceDevice::StlinkTraceDevice(std::__1::shared_ptr<spdlog::logger>) in StlinkTraceDevice.cpp.o
      StlinkTraceDevice::StlinkTraceDevice(std::__1::shared_ptr<spdlog::logger>) in StlinkTraceDevice.cpp.o
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [STMViewer] Error 1
make[1]: *** [CMakeFiles/STMViewer.dir/all] Error 2
make: *** [all] Error 2`

Meaning that the function init_chipids is not present, and when looking at the include files chipid.h installed by homebrew there is no indication of such function as it is defined in the includes from the third_party library within this repo.

I download the stlink source files and attempted to compile and because of the arm64 architecture I got a bunch of error of variables loosing precision/resolution because there a lot of assigment of size_t (64bits) to a uint32_t type variable, there were a couple of those in the chipid.c source files so I casted and fixed some of those errors and I was able to compile but ran into other installation problems...

I think the main issue here is that stlink-org is dropping support support for macos so most of the issues I'm facing other people are getting too.

I'll keep trying to see if I can get an updated version of stlink to make this work

margce, 2023-11-19:
So I was able to compile the latest stlink-org libraries from sources that include the use of init_chips. So I'm now able to compile and link correctly but when running the ./stmviewer i get the following output

`./STMViewer 
[2023-11-19 17:13:33.933] [logger] [info] Starting STMViewer!
[2023-11-19 17:13:33.934] [logger] [info] Version: 0.2.0
[2023-11-19 17:13:33.934] [logger] [info] Commit hash 5801b0845df5ab0cb659f34b1c889f5e3b2a3fd6
`
I found that the GUI mainThread fails to create the GUI window when invoking glfwCreateWindow as it returns null, but the error callback (glfw_error_callback) does not print any errors 

Any ideas on how to troubleshoot this?

klonyyy, 2023-11-19:
Is it possible for you prepare a PR with the branch you're working on so that I can try running it on a virtual machine?

margce, 2023-11-20:
> Is it possible for you prepare a PR with the branch you're working on so that I can try running it on a virtual machine?

Sure, I can do that... i did a bit more research, while trying to fix some of the warning I was getting while compiling it seems some of the functions used from openGL have been deprecated from macos 10.15 onwards, so even I fix the glfwCreateWindow call not sure how to replace those other functions.

warning: 'glViewport' is deprecated: first deprecated in macOS 10.14 - OpenGL API deprecated.
warning: 'glClear' is deprecated: first deprecated in macOS 10.14 - OpenGL API deprecated.
warning: 'glClearColor' is deprecated: first deprecated in macOS 10.14 - OpenGL API deprecated.

closed 2024-02-23 by VIPQualityPost