klonyyy/MCUViewer #25: Trace View supports jlink RTT

hongshui3000, 2023-11-29:
jlink RTT is more efficient than SWO

klonyyy, 2023-11-29:
Could you elaborate and provide some more info why is it more efficient in the way SWO is used in Trace Viewer?

hongshui3000, 2023-11-30:
[image]
https://www.segger.com/products/debug-probes/j-link/technology/about-real-time-transfer/

Candas1, 2023-11-30:
I am not sure you can compare RTT and SWO like that for this use case.
RTT is basically working like the Variable Viewer, you are polling the memory in an asynchronous way.
With SWO you are pushing the data out in a synchronous way.

klonyyy, 2023-11-30:
I don't think it would be better than SWO **in the way it is used in Trace Viewer**. This is not only about the update rate, but also the way the data is sent compared to RTT. SWO is synchronous, meaning you do not wait for the debugger to poll for RAM data. When you make a register write a sample is created with a realtive timestamp attached to it and then sent using SWO. This is why it is a great tool for measuring performance of functions/interrupts, since you can be sure about the time each sample was created (as opposed to Variable Viewer or RTT which are asynchronous and you only see the timepoints the RAM memory was sampled by the debug probe).

hongshui3000, 2023-11-30:
RTT does require polling to get data, but SWO requires the MCU to put the characters that need to be output into the register and wait for the transmission to complete, which may block the program during this period. The biggest difference between RTT and SWO is that the sending buffer becomes larger. RTT can also add time information to the data frame that needs to be output. Of course, the processing will be more complicated than SWO.

klonyyy, 2023-11-30:
Ok, but again this has a more of a Variable Viewer-like application than Trace Viewer. You have to know that Trace Viewer is not about sending strings, but rather utilize the bus as efficiently as possible with single bytes used as flags to describe logic levels. And since RTT is also implemented in target software it is less convenient, more intrusive, and it is impossible to be more efficient than SWO in this task since the timestamps are added on the level of hardware peripheral, not software.

closed 2023-12-01 by hongshui3000