klonyyy/MCUViewer #27: Can't update variable address

Eplankton, 2023-12-14:
I have an elf file compiled under arm-gcc, and there's a global variable `extern "C" volatile uint32_t  os_ticks` in the code, however when i try the viewer by the instructions, after click the update button and a black cmd window showed off, nothing changed, so I check the disassmeble of elf and didn't find any symbol like `os_ticks`, is that a problem?

Eplankton, 2023-12-14:
Already solved. Please add a description in the README that tells everybody the GDB version must >= 12.1

klonyyy, 2023-12-15:
Actually it's already in the README, maybe I should put it globally (now it's only in Windows section) or add a check in the STMViewer itself, which seems the best option.

closed 2023-12-16 by klonyyy