klonyyy/MCUViewer #29: Problem Updating Variable

Amitkor12, 2024-01-31:
Hello, I want to express my gratitude for creating such a fantastic application that proves to be highly useful for robotics. I have downloaded the application and carefully followed the instructions, but I encountered an issue where the variables do not get updated, and the addresses are not displayed as expected.
I would appreciate any suggestions, hints, or insights you could provide regarding the issue I'm facing with.

Thank You.

klonyyy, 2024-01-31:
Hello, could you please provide more details? For example a project file and the *.elf file you're trying to parse. I'll try to reproduce  the problem locally. Please make sure the GDB version is correct as well (#27)

Amitkor12, 2024-02-01:
[firmware.zip](https://github.com/klonyyy/STMViewer/files/14120430/firmware.zip)
This is the elf file i am using.
The gdb version is also update to latest version 14.2

klonyyy, 2024-02-01:
I seems to work fine on my end:
[image]

Are there any particular variables you have troubles importing? 
Please make sure your path does not contain any spaces (#27)

Amitkor12, 2024-02-01:
Thanks for the help.
First i tried this on windows 11. Here i did not get any response in updating variables , but with the same file on linux i was able to update the variables.
[image]
this is on linux ubuntu 20.04 lts.

klonyyy, 2024-02-01:
This is strange, as my screenshot is from Windows 10. If you've got some spare time you can check for the spaces in your *.elf path. I'm curious if it's a bug.

Amitkor12, 2024-02-02:
I checked the path of the *.elf file, and indeed, there was a space between the names. Now, after the correction, it works fine. Thanks for your support.

closed 2024-02-02 by Amitkor12