klonyyy/MCUViewer #34: Unlimited data points enhancement

Eplankton, 2024-02-14:
For now, STMViewer supports only 20000 points as max, so if a high freq signal comes, it will only show a very small piece of data point stream.

klonyyy, 2024-02-14:
I agree that sometimes 20000 points is too little. I can increase the limit, however the GUI performance will degrade the more points are shown on a single plot, since there is no display optimization implemented on the ImPlot side. I'm adding this issue to TODO list.

Eplankton, 2024-02-15:
> I agree that sometimes 20000 points is too little. I can increase the limit, however the GUI performance will degrade the more points are shown on a single plot, since there is no display optimization implemented on the ImPlot side. I'm adding this issue to TODO list - ideally I'd like to do it when the display optimization is implemented.

Maybe we can just add a feature to continuously output unlimited data point to a .csv file(automatically, don't have to click the `save as csv` button), without display it.

Eplankton, 2024-02-22:
> Ah ok, so it would be like a logger, without the visual representation? Seems like an interesting idea, let me add it to future/planned features. I'd need some more time to think it through.

Yes, as a logger, so I can use other tools like python to process, analyze or visualize these data.

closed 2024-10-03 by klonyyy