klonyyy/MCUViewer #39: Quick Start not clear

42Bastian, 2024-02-22:
The QuickStart is not very clear on how to use this tool.
I can start it, I can select a variable.
But how to "download"? My application is flashed and I use the board ST-Link (nucleo-h563zi).

Maybe a more elaborate description will help.

thx

Candas1, 2024-02-22:
I made this video some time ago
https://youtu.be/mUbeMSEr2H8?si=hPyfwL_eDxlTt6PM

klonyyy, 2024-02-22:
Hello, what do you mean by "download"? STMViewer's variable viewer is used to connect to a running target and reading the memory values off addresses based on the application *.elf file. 

Fell free to suggest improvements as some things might be obvious for me and maybe I'm not aware of thing that could be unclear. 

@Candas1 thanks for the video! I'm also planning on making an official one sometime soon ;)

42Bastian, 2024-02-22:
For the QuickStart: It is not clear it directly connects to a running target.

Ok, the big red button was misleading. I clicked on it, and it either says "STOPPED STLink not found" (true, I have STCubeProgrammer connected) or "STOPPED STM32 target not found!"

Could it be, it does not work with STM32H563ZI? I tried CubeMonitor and this one works (though I'd prefer this simpler interface).

klonyyy, 2024-02-22:
Could you attach the logs (either `~/STMViewer/logs/` or `C:\Users\<USER>\AppData\Roaming\STMViewer\logs\` on Windows) ? Did you built the STMViewer yourself or used the attached installers?

klonyyy, 2024-02-22:
This is in case in which no other program uses the STLink? Could you make sure, for example by repluging the stlink before trying again? Also could you check (for example using STM32 Cube programmer) if these params match your STM32: 

```
# Chip-ID file for STM32H5xx device
#
dev_type STM32H5xx
ref_manual_id 0481
chip_id 0x484                // STM32_CHIPID_H5xx
flash_type L5_U5_H5
flash_size_reg 0x08fff80c
flash_pagesize 0x2000        // 8 KB
sram_size 0xa0000            // 640 KB
bootrom_base 0x0bf80000
bootrom_size 0x8000          // 32 KB
option_base 0x0
option_size 0x0
flags dualbank
```

these are the params from the stlink library used by the STMViewer. Maybe there's some issue with the STM32H5 config.

42Bastian, 2024-02-22:
I am checking, but having trouble finding the addresses in the STM manuals.
At least the chip_id, flash page and total size and sram_size fit.

42Bastian, 2024-02-22:
In STMCubeMonitor I had to set the access port to 1 instead 0. But do not see if this can be done via .chip file.

klonyyy, 2024-02-22:
Are you working with SWD or JTAG?

42Bastian, 2024-02-22:
> Are you working with SWD or JTAG?

SWD

42Bastian, 2024-02-22:
Ok, I give up. Thanks for replies. I tried FOSS stlink tool, but it crashes.

closed 2024-02-22 by 42Bastian