klonyyy/MCUViewer #40: Display (parts of) arrays

wimalopaan, 2024-02-23:
It would be convenient to have an option to display the contents of an array (e.g. like the output of a fft) and update that continously. Sometimes it would also be good to only display a part of that array e.g. an index-range.

klonyyy, 2024-02-23:
Do you mean it would be nice to have a way of quickly importing an array, or rather displaying a range of an array as a table view?

wimalopaan, 2024-02-23:
A table-view would be good, but I had an updating graphical view (e.g. a bar-graph) in mind.
And it would be very nice if it would be possible to import the whole array at once.

wimalopaan, 2024-02-23:
Another option would be so save the array-data in a trace-file periodically triggered by some event / variable-change.

klonyyy, 2024-02-23:
So considering options that are available right now, the main problem is that the import and placing the variables on the plot takes very long to set up, am I right?

wimalopaan, 2024-02-23:
Not only that, but I'm looking for a way to plot an array as a whole: e.g. the array contains some fft-values (or the amplitudes of the frequencies), I want to see the spektrum.

wimalopaan, 2024-02-25:
Yes, a bar graph is an option but specifying the indixes is tedious (e.g. a 512 element array).

Maybe a another option would be to write the whole array to a file (together with some other variable that represents to time).

klonyyy, 2024-02-25:
Thanks for clarification. The functionality you're porposing seems to be connected to #34. I'll think about the optimal way to implement it.

wimalopaan, 2024-02-25:
Yes, I overlooked #34 . Thank you!

wimalopaan, 2024-05-17:
For the bar-graph of arrays: do you see any change to implement this?

klonyyy, 2024-05-17:
Unfortunately my current priority is to merge the devel branch with J-Link HSS feature to main. After that I plan on adding less anticipated features. If you're not able to help directly by implementing the feature, It would be great if you could provide a detailed description of how you'd see this feature. I wouldn't like to implement FFT-type plots using bargraphs if they are not designed for such usecases. You could check out ImPlot for the type of plot that would suit this best.

closed 2024-10-03 by klonyyy