klonyyy/MCUViewer #41: Custumization capabilities

Candas1, 2024-02-23:
I remember I came across [this](https://github.com/markisus/motor_sim?tab=readme-ov-file#bldc-and-pmsm-simulator) simulation a long time ago that seems to use the same UI library:
[image]

This got me thinking, do you want to keep this tool generic or are you thinking about adding customizations capabilities in the future ?

A few examples I have in mind:
- being able to provide a short description for a variable
- being able to provide a unit for a variable
- new UI elements - sliders,gauges,checkboxes(would require min/max/step to be set by the user)
- combobox/radio buttons (might be complicated if it needs descriptions to match values)
- being able to resize/move the plots
- being able to group plots/tables and use tabs

klonyyy, 2024-02-25:
So I would like the STMViewer to stay relatively generic, meaning I don't want it to go the way CubeMonitor went. From my perspective such layouts like the one above are very application specific and this would make the STMViewer usage a lot harder as well. Moreover it is designed for the debug interface, and usually this would mean only the developer would use it, as end-users don't have the debug probe and the access to the debug port. That being said, I was thinking of adding:

- plot collections in tabs so that the whole view can be quickly modified 
- special variables for results of math operations on real variables

VIPQualityPost, 2024-02-27:
However, if we change graphics libraries, it could also be possible to support macOS among others... 😁  the screenshot @Candas1 posts reminds me of Dear Imgui like used on ngscopeclient

dzid26, 2024-05-03:
Simple reordering of the plots would be nice. 
Currently, I believe plots are in order of when they were Added. I don't think it is possible to drag them around.

closed 2024-05-23 by klonyyy