klonyyy/MCUViewer #43: My time base is wrong

Trampler60, 2024-04-02:
This is 50 Hz signal directly from ADC. As you see my there is a problem about timing.

[image]
[image]

Trampler60, 2024-04-03:
"Could you verify the timing with systick?" How can i do this, can you explain further?

STM32F103C8
[image]

klonyyy, 2024-04-04:
Sure, you could write a value to the CH0 in the SysTick 1kHz interrupt so that we are sure the problem is on STMViewer's side. You can write any value, just make sure it changes every interrupt so that the steps indicate the time period we want to examine.

Trampler60, 2024-04-04:
I also tried to increase systick freq. I was able to increase it 10 times and it seems ok. But when ı tried 100 times it broke again. Am i stuck at speed limit? I tried prescaler between 36 to 72 this means a period of 1-2 microseconds . 
Why does the problem occur when the systick period is 10 microseconds? 

[image]

normal settings: HAL_SYSTICK_Config(HAL_RCC_GetHCLKFreq()/1000); //1000
[image]

HAL_SYSTICK_Config(HAL_RCC_GetHCLKFreq()/10000); //10000
[image]

HAL_SYSTICK_Config(HAL_RCC_GetHCLKFreq()/100000); //100000
[image]

Trampler60, 2024-04-04:
I think you are right about saturation.  I am using STMViewer 0.4.0. Maybe ı need better STLİNK.
[image]

klonyyy, 2024-04-05:
Yes this is possible. I can recommend the ST-Link V3 minie. 

Why did you have to check the frames in Cube IDE? Is there a problem with the trace indicators on the left panel in STMViewer? A full screen screenshot would be helpful in the case you're suspecting a saturation.

klonyyy, 2024-04-22:
Closing due to inactivity. If you see a similar issue please reopen.