klonyyy/MCUViewer #45: Counter Plotting

Mezamoha, 2024-04-24:
Hello,

I would like to thank you for the great software.

I have a question :

I tried to profile some function and counters and it works well 
for example

```
while(1){
ITM->PORT[0].u32 = size;
size++;
}
```

And the plot is as expected 

[image]

but when i tried to plot a cycle counter 

```
while(1){
ITM->PORT[0].u32 = DWT->CYCCNT;
}
```

i get a fixed value so i'm not able to see the progression of this counter

[image]

I would like to know if there is any solution or method to be able to get the correct plotting

Thanks a lot

klonyyy, 2024-04-24:
Hello, this might be because DWT registers are altered during trace setup. The DWT might be used by the ITM peripheral (as far as  I remember it is not used currently). Do you set up the counter by yourself in your code? Why do you want to display the cycles count?

Mezamoha, 2024-04-24:
Thank you for your answer

In fact i'm interested in monitoring real-time counters such as LSUCNT, CPICNT, EXCCNT with Stmviewer. Here is my c code that i'm using with Stmviewer. The purpose is to know how those counters are impacted by our application.
Is it possible to monitor these counters with Stmviewer in real-time ?

```
#include "core_cm4.h" 

void DWT_Init(void) {
    CoreDebug->DEMCR |= CoreDebug_DEMCR_TRCENA_Msk;
    DWT->CTRL |= (DWT_CTRL_CYCCNTENA_Msk | DWT_CTRL_CPIEVTENA_Msk | DWT_CTRL_LSUEVTENA_Msk | DWT_CTRL_EXCEVTENA_Msk);
}

volatile void ITM_SendData(int port, unsigned int data) {
    while (ITM->PORT[port].u32 == 0);
    ITM->PORT[port].u32 = data;
}

void Send_DWT_Counters(void) {
    unsigned int cycles = DWT->CYCCNT;
    unsigned int cpi_count = DWT->CPICNT;
    unsigned int lsu_count = DWT->LSUCNT;
    unsigned int exc_count = DWT->EXCCNT;

    ITM_SendData(0, cycles);
    ITM_SendData(1, cpi_count);
    ITM_SendData(2, lsu_count);
    ITM_SendData(3, exc_count);
}

int main(void) {
    ITM_Init();
    DWT_Init();

    while (1) {
        Send_DWT_Counters();
    }
}
```

klonyyy, 2024-04-24:
The main problem is that STMViewer's Trace Viewer module uses the ITM and DWT. If you do not need a synchronous output of the data I'd use the Variable Viewer and just assign these counter values to some global variables and plot them. This way you'll be able to read the counters in real time. 

The Trace Viewer aims to be automatic (it sets up the registers by itself), this is why interfering with ITM and DWT in your application results in faulty output.

klonyyy, 2024-04-24:
Actually the Trace Viewer does not utilize these counters. You can do it by hand in your application, just the display part in STMViewer should be done using VariableViewer, as it does not change the ITM settings.

Mezamoha, 2024-04-25:
Great.
Thanks a lot 
Have a good day

closed 2024-04-25 by klonyyy