klonyyy/MCUViewer #49: CPU/GPU usage when idle

dzid26, 2024-05-03:
Windows 11. 
There is constant small CPU/GPU usage even when acquisition is **stopped** and app is **minimized**. 

Can we pause the rendering at least under those conditions?

klonyyy, 2024-05-03:
Could you show me the circumstances under which these usages take place? This is my usage when STMViewer is minimized:
[image]

dzid26, 2024-05-03:
In general the usage is not big (usually 2%) but it jumps to 5% periodically. 
Examples. 
(When recording is not active, the effect on GPU is slightly larger. )

https://github.com/klonyyy/STMViewer/assets/841061/79d86b5d-a6b1-4129-a473-e70b54eea0b6

https://github.com/klonyyy/STMViewer/assets/841061/58e69c3a-afc5-450b-8a8f-20b8264e1a1d

I am expecting there could be 0% when stopped. 
But if imgui is not efficient about mouse events, etc.., then at least maybe we can pause it when minimized. 

Thanks

klonyyy, 2024-05-04:
Please let me know if the latest commit solved the issue. Thanks for bringing it out!

dzid26, 2024-05-04:
Strangely, today, I couldn't reproduce 5% CPU spikes on any version. 

Version 0.4 & 0.5:
- When stopped CPU was mostly 0% and GPU mostly  4%.
- When minimized CPU was mostly at 0% and GPU at 0.5% average

With `GLFW_FOCUSED`:
- CPU and GPU are both 0% when not in focus.  Great.

But the cursors don't refresh when hovering.  :( 

Since the issue doesn't seem as bad as I thought, perhaps we can check for "minimized" only?  

This works:
```
if (glfwGetWindowAttrib(window, GLFW_ICONIFIED))
```

klonyyy, 2024-05-04:
Yeah, I was thinking of the GLFW_ICONIFIED however this does not seem to limit the rendering whenever the windows is overlayed by another window (not minimized explicitly). Let's stick with it and see if someone complaints about the high CPU/GPU usage after this fix.

dzid26, 2024-05-05:
Actually, we also could add this in the while loop:
```
		if (glfwGetWindowAttrib(window, GLFW_FOCUSED) || (tracePlotHandler->getViewerState() == PlotHandlerBase::state::RUN) || (plotHandler->getViewerState() == PlotHandlerBase::state::RUN))
			glfwSwapInterval(1);
		else
			glfwSwapInterval(4);
```

1% instead of 4% GPU usage with visible window.

klonyyy, 2024-05-05:
Would you be willing to implement and test that in a PR to the devel branch?

closed 2024-05-06 by dzid26