klonyyy/MCUViewer #54: Time based X axis scaling

dzid26, 2024-05-03:
This is sample based:
[image]

This doesn't work
[image]

It's a nice to have...

klonyyy, 2024-05-03:
Please explain further. I'm not sure what you mean.

dzid26, 2024-05-03:
I am referring to viewport X axis [time] range . 
Currently it can be only globally controlled from Acquisition settings using number of samples. 

It would be more powerful if viewport time range was controllable from the plot right click menu - using time units and maybe with the mouse- during acquisition. 

- Currently, I believe, Autofit and Min / max are ignored on time axis - when acquisition is started. 
- Zooming in and out with the mouse only works during acquisition.  

In contrast, for Y axis the range can be changed during acquisition. Although, zooming with the mouse also doesn't.

dzid26, 2024-05-03:
The perfection would be the following:
- Mouse zooming available for X and Y when stopped _and_ during acquisition. 
- Min/max possible to adjust during stop, but not forgotten during acquisition starts, - related to https://github.com/klonyyy/STMViewer/issues/52

klonyyy, 2024-09-20:
Please reopen/react to this post if you agree this feature is needed. I'm closing for now.

closed 2024-09-20 by klonyyy