klonyyy/MCUViewer #55: Space key to start/stop acquisition

dzid26, 2024-05-03:


dzid26, 2024-05-04:
I saw your commit. 

In addition I was also looking at using `io.ConfigFlags |= ImGuiConfigFlags_NavEnableKeyboard;` (which is nice in itself for navigating toolbar menu) with `ImGui::SetKeyboardFocusHere(0);` (placed before start button).  

It creates this pretty frame around the start button, and it is visible that the button is being pressed.  But it flickers, since the focus is called all the time and interferes with toolbar navigation.

This may not be possible to solve easily. 

Also related:
https://github.com/ocornut/imgui/issues/5606#issuecomment-1833225564

klonyyy, 2024-05-05:
I think navigation using the keyboard is a secondary task, as most people using STMViewer I know, are primarily navigating with the mouse. Let's see if the issue gains more attention in the future. I'll close it for now but feel free to reopen, vote for adding such feature or contribute to implement it.

closed 2024-05-05 by klonyyy