klonyyy/MCUViewer #57: Renders at only 30fps

dzid26, 2024-05-04:
[image]

Tested with:
```
	std::string fpsStr = std::to_string(ImGui::GetIO().Framerate);
	ImGui::BeginMenu(fpsStr.c_str());
```

klonyyy, 2024-05-04:
Do we really need more? With many points on the viewport the interface starts to get laggy - I would expect it to only get worse with higher fps. I think this should stay as it is, at least to the moment the points displaying algorithm is optimized (so that it displays the least amount of points able to represent the shape of the waveform).

dzid26, 2024-05-05:
It does look much better at full refresh rate.  It stutters often, just like any Windows windowed app, but I think that is dwm related. 

I don't get why FPS eventually drops to 30fps when there is many points accumulated (not even displayed, just accumulated) - When that happens the GPU usage is still low (13%) and single core is only at about 30%. (I find fps counter helpful to identify when this happens.)

When acquisition is restarted, FPS goes back up again.  

Overall, I prefer that to always have it capped to half refresh rate..

dzid26, 2024-05-05:
Btw, if I put 100 max points and 100 view points 
[image]
why is there many more points visible after zooming out ?
[image]

I might be misinterpreting this, but I thought that if the values are matching there shouldn't be more points stored than there is visible on the rolling view.

klonyyy, 2024-05-07:
This is a bug (#62). Just fixed it - thanks for reporting it.

klonyyy, 2024-05-07:
> It does look much better at full refresh rate. It stutters often, just like any Windows windowed app, but I think that is dwm related.
>
> I don't get why FPS eventually drops to 30fps when there is [...]

This is something that I'd have to investigate in more detail. Currently it's not my top priority so I will not be to look into it soon.  Feel free to work on it yourself though and if you do make sure to create a PR for it :)

closed 2024-05-23 by klonyyy