klonyyy/MCUViewer #63: How to Run the program

keremekinci, 2024-05-13:
Hello,
  I encountered some problems while trying to install the program, 
  cmake -G "MinGW Makefiles" -B build
 After the command, I enter the folder with cd build and run the makefile all command, 
 but  when i use "make all" command 
[ 2%] Linking CXX static library libnfd.a
Error running link command: no such file or directorymake[2]: *** [third_party/nfd/src/CMakeFiles/nfd.dir/build.make:99: third_party/nfd/src/libnfd.a] Error 2
make[1]: *** [CMakeFiles/Makefile2:169: third_party/nfd/src/CMakeFiles/nfd.dir/all] Error 2
make: *** [makefile:156: all] Error 2

 I am getting this  error how do I solve it?

klonyyy, 2024-05-14:
Hello, which platform are you on? Could you check the readme file about building on the devel branch?

klonyyy, 2024-05-23:
Closing due to inactivity.

closed 2024-05-23 by klonyyy