klonyyy/MCUViewer #65: Allow relative elf file path in cfg file

dzid26, 2024-05-15:
I wanted to add project file to revision control, but it contains my absolute path.

Related:
https://github.com/klonyyy/STMViewer/blob/4d22bcdeb6d0a01148aff162177f075b49920258/test/STMViewer_test/STMViewer_project/STMViewer_test.cfg#L2

dzid26, 2024-05-17:
But does it work? 

I tried:
```
[elf]
file_path = .\.pio\build\ServoCAN_release\firmware.elf
```

or:
```
[elf]
file_path = .pio\build\ServoCAN_release\firmware.elf
```

and it fails when _updating variable addresses_. 
[image]

klonyyy, 2024-05-18:
Relative means to the place the binary is run from. So you have to make sure the directory you're referring to is in the binary folder, otherwise you have to step out of it using ../ (you can see the example in my commit - I've checked it on both Windows and Ubuntu and it works fine)

dzid26, 2024-05-18:
So this only works for your example. 

The path should be relative to cfg file location.  Can you reopen the issue?

klonyyy, 2024-05-18:
Now I understand what you meant. Please provide a full issue description next time so that we're both on the same page.

closed 2024-06-09 by klonyyy