klonyyy/MCUViewer #73: STMViewer and STM32H7

folny, 2024-06-20:
### Version/Branch:

5.0

### Operating system:

windows 11

### Debug probe:

ST-Link v2

### Details:

Hello

Has anyone tried STMViewer with an STM32H7 processor?. In my case, Trace Viewer doesn't want to work. I have this particular board https://www.aliexpress.us/item/3256806446021431.html?spm=a2g0o.order_list.order_list_main.20.47861c247DAPaT&gatewayAdapt=glo2usa4itemAdapt

klonyyy, 2024-06-20:
Hello, please provide more details, for example a screenshot, and the latest log file (located C:\Users\<user>\AppData\Roaming\STMViewer\logs).

folny, 2024-06-20:
My test code and cpu clock
[image]

while (1) {
        ITM->PORT[0].u8 = 0xaa;
		    status ^= 1;
				HAL_GPIO_TogglePin(LED_GPIO_Port,LED_Pin);
			  ITM->PORT[0].u8 = 0xbb;
	      HAL_Delay(1);
    }

folny, 2024-06-20:
Var Viewer OK
[image]

folny, 2024-06-20:
Trace Viewer not work
[image]

folny, 2024-06-20:
Logs

[2024-06-20 21:16:40.577] [logger] [info] Starting STMViewer!
[2024-06-20 21:16:40.577] [logger] [info] Version: 0.5.0
[2024-06-20 21:16:40.577] [logger] [info] Commit hash f65dd138e0127e46cb946a40c1d7dc4a7aa3d073
[2024-06-20 21:16:40.749] [logger] [info] GDB version: 121
[2024-06-20 21:16:41.061] [logger] [info] Unable to match requested speed 24000 kHz, using 4000 kHz
[2024-06-20 21:16:41.063] [logger] [info] STM32H74x_H75x: 128 KiB SRAM, 2048 KiB flash in at least 128 KiB pages.
[2024-06-20 21:16:41.063] [logger] [info] STLink serial number 066BFF323532534157163257
[2024-06-20 21:16:45.239] [logger] [info] Adding variable: status
[2024-06-20 21:16:45.239] [logger] [info] Adding plot: new plot0
[2024-06-20 21:16:45.239] [logger] [info] Adding series: status
[2024-06-20 21:16:45.239] [logger] [info] Adding trace plot: CH0
[2024-06-20 21:16:45.239] [logger] [info] Adding trace plot: CH1
[2024-06-20 21:16:45.239] [logger] [info] Adding trace plot: CH2
[2024-06-20 21:16:45.239] [logger] [info] Adding trace plot: CH3
[2024-06-20 21:16:45.239] [logger] [info] Adding trace plot: CH4
[2024-06-20 21:16:45.239] [logger] [info] Adding trace plot: CH5
[2024-06-20 21:16:45.240] [logger] [info] Adding trace plot: CH6
[2024-06-20 21:16:45.240] [logger] [info] Adding trace plot: CH7
[2024-06-20 21:16:45.240] [logger] [info] Adding trace plot: CH8
[2024-06-20 21:16:45.240] [logger] [info] Adding trace plot: CH9
[2024-06-20 21:16:45.240] [logger] [info] Project config path: D:/Projekty/STM32H7_Metal_Detector/Firmware/STM32H7_NRF24_LCD_Data_Test/Data.cfg
[2024-06-20 21:16:45.278] [logger] [info] Unable to match requested speed 24000 kHz, using 4000 kHz
[2024-06-20 21:16:45.280] [logger] [info] STM32H74x_H75x: 128 KiB SRAM, 2048 KiB flash in at least 128 KiB pages.
[2024-06-20 21:16:45.280] [logger] [info] STLink serial number 066BFF323532534157163257
[2024-06-20 21:16:47.719] [logger] [info] Start trace clicked!
[2024-06-20 21:16:47.749] [logger] [info] Unable to match requested speed 24000 kHz, using 4000 kHz
[2024-06-20 21:16:47.750] [logger] [info] STM32H74x_H75x: 128 KiB SRAM, 2048 KiB flash in at least 128 KiB pages.
[2024-06-20 21:16:47.752] [logger] [info] Trace frequency 2000000
[2024-06-20 21:16:47.752] [logger] [info] Trace prescaler 31
[2024-06-20 21:16:47.752] [logger] [info] Trace channels mask 1
[2024-06-20 21:16:47.754] [logger] [info] Starting reader thread!
[2024-06-20 21:16:50.333] [logger] [error] No trace registered for 2s!
[2024-06-20 21:16:50.333] [logger] [error] Trace invalid, stopping!
[2024-06-20 21:16:50.334] [logger] [info] Trace stopped.
[2024-06-20 21:16:50.334] [logger] [info] Closing trace thread false
[2024-06-20 21:17:14.533] [logger] [info] Unable to match requested speed 24000 kHz, using 4000 kHz
[2024-06-20 21:17:14.533] [logger] [info] STM32H74x_H75x: 128 KiB SRAM, 2048 KiB flash in at least 128 KiB pages.
[2024-06-20 21:17:14.533] [logger] [info] STM32H74x_H75x: 128 KiB SRAM, 2048 KiB flash in at least 128 KiB pages.
[2024-06-20 21:17:20.718] [logger] [info] Unable to match requested speed 24000 kHz, using 4000 kHz
[2024-06-20 21:17:20.719] [logger] [info] STM32H74x_H75x: 128 KiB SRAM, 2048 KiB flash in at least 128 KiB pages.
[2024-06-20 21:17:20.721] [logger] [info] STM32H74x_H75x: 128 KiB SRAM, 2048 KiB flash in at least 128 KiB pages.
[2024-06-20 21:17:27.912] [logger] [info] Start trace clicked!
[2024-06-20 21:17:27.946] [logger] [info] Unable to match requested speed 24000 kHz, using 4000 kHz
[2024-06-20 21:17:27.948] [logger] [info] STM32H74x_H75x: 128 KiB SRAM, 2048 KiB flash in at least 128 KiB pages.
[2024-06-20 21:17:27.950] [logger] [info] Trace frequency 2000000
[2024-06-20 21:17:27.950] [logger] [info] Trace prescaler 31
[2024-06-20 21:17:27.950] [logger] [info] Trace channels mask 1
[2024-06-20 21:17:27.952] [logger] [info] Starting reader thread!
[2024-06-20 21:17:30.573] [logger] [error] No trace registered for 2s!
[2024-06-20 21:17:30.573] [logger] [error] Trace invalid, stopping!
[2024-06-20 21:17:30.574] [logger] [info] Trace stop

klonyyy, 2024-06-20:
Have you enabled SWD + trace (SWO) setting in cube IDE? 
[image]

folny, 2024-06-22:
Yes, this is fine, without this setting Var Viewer probably wouldn't work either.
[image]

klonyyy, 2024-06-22:
Have you checked if the SWO outputs any data to the ST-Link on the SWO pin? 

Var Viewer works only with SWCLK, SWDIO and GND. SWO is needed additionally for Trace Viewer.

folny, 2024-06-22:
I have not tried to check that, I will try to connect the oscilloscope to the SWO pin today and let you know if there is any data on the pin.

folny, 2024-06-23:
I found that I have no data on the SWO pin. Please check my .ioc file if everything is set correctly.
[STM32H743_STMViewer_Test.zip](https://github.com/user-attachments/files/15943841/STM32H743_STMViewer_Test.zip)

klonyyy, 2024-06-23:
The *.ioc seems to be correct. Have you checked the SWO signal directly at the MCU pin? Are you sure there is nothing connected to it that could interfere with the signal?

folny, 2024-06-24:
Today I tested the second STM32H7 this time I have data on pin SWO but STMViewer does not show any data.

[image]
[image]

klonyyy, 2024-06-24:
Ok this is an improvement. Could you try different prescaler values? If you're not using an original stlink there may be some problems with maximum SWO pin bandwidth. Could you try an official stlink?

folny, 2024-06-24:
I use ST-Link with a NUCLEO-F446RE development board

https://sk.farnell.com/en-SK/stmicroelectronics/nucleo-f446re/dev-board-arduino-mbed-nucleo/dp/2491978

klonyyy, 2024-06-25:
Have you tried modifying the prescaler?

folny, 2024-06-25:
Yes, I tried different values ​​but nothing is displayed. With cpu STM32f446 it works beautifully, only with H7 I have a problem.

klonyyy, 2024-06-25:
Could you try connecting the oscilloscope so that you see proper frames? It seems you only get error frames, so maybe the problem is in the electrical connection somehow? The scope last screen you've sent me seems to have very bad edges and overall signal quality.

folny, 2024-07-02:
Today I tested the same code on the STM32F446 NUCLEO and the data on the SWO pin is fine and looks like this.

[image]

folny, 2024-07-02:
The next test is on the H7 and there is a much higher frequency on the SWO pin, even if I change the prescaler in STMViewer to a higher value, no changes are reflected.

[image]

folny, 2024-07-03:
So today I searched on the internet what could be the problem with the SWO and I found this configuration that helped me and the STM32H7 SWO is now working properly :-) 

https://gist.github.com/tstellanova/107cb66cd0599fa20cca1fce9556468c

klonyyy, 2024-07-04:
Thank you for the research! Is it possible for you to narrow down the whole SWO setup to a particular register write(s) that make it work with STMViewer?

folny, 2024-07-05:
Yes, STMViewer works fine with this setup.

klonyyy, 2024-07-05:
So I meant like, are all these register writes indispensable? Or you could test which one is really necessary?

folny, 2024-07-05:
I haven't tried to change anything yet, I will try to make some changes and add more information.

closed 2024-09-20 by klonyyy