klonyyy/MCUViewer #74: STM32H573 target not found

antoinh, 2024-06-21:
### Version/Branch:

0.5 release

### Operating system:

Windows 10

### Debug probe:

ST-LINK V3

### Details:

Hello, 

First thanks for the  tool and sharing it with the community!! It is awesome.

I used the precompiled windows executable, I installed it and I was able to make the variable viewer work on an STM32G0B without any issues.

When switching to an to an STM32H573 however I am unable to connect and I get a Target not found error when cliking on the start button. Trace SWO is activated.

Here is the content of the log file

[2024-06-21 14:12:43.941] [logger] [info] Starting STMViewer!
[2024-06-21 14:12:43.945] [logger] [info] Version: 0.5.0
[2024-06-21 14:12:43.945] [logger] [info] Commit hash eb3030ff291405aaf3abc42b1ef2cdbd2f49b163
[2024-06-21 14:12:44.455] [logger] [warning] Failed to read GDB version (from PATH). Retrying on the distributed GDB...
[2024-06-21 14:12:44.464] [logger] [info] GDB version: 102
[2024-06-21 14:12:58.421] [logger] [error] Can not connect to target. Please use 'connect under reset' and try again
[2024-06-21 14:12:58.421] [logger] [info] STLink serial number 002800293038511834333935
[2024-06-21 14:13:02.490] [logger] [info] Project elf file path: F:/Projets/MicroRing/Design/Firmware/Detection/cmake-build-debug/MicroRing_Detection.elf
[2024-06-21 14:13:22.685] [logger] [info] Start clicked!
[2024-06-21 14:13:22.779] [logger] [error] Can not connect to target. Please use 'connect under reset' and try again
[2024-06-21 14:13:22.780] [logger] [error] Can not connect to target. Please use 'connect under reset' and try again
[2024-06-21 14:13:24.518] [logger] [info] Start clicked!
[2024-06-21 14:13:24.608] [logger] [error] Can not connect to target. Please use 'connect under reset' and try again
[2024-06-21 14:13:24.609] [logger] [error] Can not connect to target. Please use 'connect under reset' and try again
[2024-06-21 14:13:34.701] [logger] [info] Start clicked!
[2024-06-21 14:13:34.807] [logger] [error] Can not connect to target. Please use 'connect under reset' and try again
[2024-06-21 14:13:34.808] [logger] [error] Can not connect to target. Please use 'connect under reset' and try again
[2024-06-21 14:13:36.068] [logger] [info] Start clicked!
[2024-06-21 14:13:36.143] [logger] [error] Can not connect to target. Please use 'connect under reset' and try again
[2024-06-21 14:13:36.144] [logger] [error] Can not connect to target. Please use 'connect under reset' and try again
[2024-06-21 14:13:36.286] [logger] [info] Start clicked!
[2024-06-21 14:13:36.352] [logger] [error] Can not connect to target. Please use 'connect under reset' and try again
[2024-06-21 14:13:36.353] [logger] [error] Can not connect to target. Please use 'connect under reset' and try again
[2024-06-21 14:13:36.486] [logger] [info] Start clicked!
[2024-06-21 14:13:36.564] [logger] [error] Can not connect to target. Please use 'connect under reset' and try again
[2024-06-21 14:13:36.565] [logger] [error] Can not connect to target. Please use 'connect under reset' and try again
[2024-06-21 14:13:48.318] [logger] [info] Start clicked!
[2024-06-21 14:13:48.411] [logger] [error] Can not connect to target. Please use 'connect under reset' and try again
[2024-06-21 14:13:48.412] [logger] [error] Can not connect to target. Please use 'connect under reset' and try again
[2024-06-21 14:13:51.053] [logger] [info] Exiting trace plot handler thread
[2024-06-21 14:13:51.070] [logger] [info] Exiting GUI main thread
[2024-06-21 14:13:51.502] [logger] [info] Closing STMViewer!

Any idea what could be wrong? 

Thanks !

klonyyy, 2024-06-22:
Hello, I think there were problems with H5 microcontrollers (https://github.com/klonyyy/STMViewer/issues/39). I do not have any H5 evals on my hand right now, but I can try to help you debug the issue. Could you try running the [st-link](https://github.com/stlink-org/stlink) software and try if it connects? Maybe this is only a matter of updating stlink library on my side.

antoinh, 2024-06-25:
I am not very familiar with the stlink tools but I can connect fine using the STM32CubeProgrammer utility.

I did try to use the precompiled binaries for windows for stlink but couldn't get them to work on any of my ST boards so I am probably doing something wrong.

stlink-info --probe

returns no output 

Thanks for the help :)

 
[image]

[image]

klonyyy, 2024-06-25:
Yeah I've just checked - you have to copy a libusb-1.0.dll to make it work: 
[image]

You can take it from STMViewer install folder, or download from the internet.

antoinh, 2024-06-25:
Ah ! Good catch. Can't seem to connect still though :( 

[image]

Sorry didn't have much time to troubleshoot this today, i'll try again tomorrow

antoinh, 2024-06-25:
if that can help when using cube programmer I get this warning message. Not quite sure how I could select the access port with the direct stlink utility

[image]

tonyh1988, 2024-06-26:
unfortunately it seems these newer devices (cortex33) are not supported yet by stlink 

https://github.com/stlink-org/stlink/issues/1398

I'm pretty sure it has a lot to do with the access port (needs to be AP1) but that option is no present in the open source stlink:

https://community.st.com/t5/stm32cubeide-mcus/unable-to-flash-nucleo-h563zi-with-stm32cubeide-v1-14/m-p/618423/highlight/true#M22695

tonyh1988, 2024-07-02:
There might be a way to use the stm32cube programmer c++ API instead it seems more up to date. I didn't know this thing existed until now when I read the manual: 

https://www.st.com/resource/en/user_manual/um2237-stm32cubeprogrammer-software-description-stmicroelectronics.pdf

To have the documentation you need to download cube programmer and everything is under the api folder. 
How feasible do you think it would be to try and use this instead in your software? I could give it a go but you know the codebase far more than I right now

klonyyy, 2024-07-02:
I didn't know they provided a C++ API! This seems to be a solution to the problem if all the needed functions are available. I think it lacks the SWO (trace) readout API, but for a Variable Viewer this might be enough. If you'd like we can connect and I can tell you what needs to be implemented to make at least a quick PoC ;)

klonyyy, 2024-09-20:
After more consideration I don't think I'll go in the cube monitor API support now. It's too much hassle to maintain in parallel with stlink library that is currently used (we need trace support unfortunately. If you'd like to use the H5 series please go fo JLink for now.

closed 2024-09-20 by klonyyy