klonyyy/MCUViewer #80: Can't install on ubuntu through parallels on Mac with Apple M1

cosmikwolf, 2024-09-20:
### Version/Branch:

1.0.0

### Operating system:

Ubuntu 22.04

### Debug probe:

Jlink plus

### Details:

I downloaded the latest .deb file from the releases page, and I tried to install it on my parallels VM running ubuntu 22.04 and I get this error. I haven't ever hit this before, so not sure what to do. 

```
parallels@ubuntu-vm ~> sudo apt install ./MCUViewer-1.0.0-Linux.deb
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'mcuviewer:amd64' instead of './MCUViewer-1.0.0-Linux.deb'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 mcuviewer:amd64 : Depends: libglfw3:amd64 but it is not installable or
                            libglfw3-wayland:amd64 but it is not installable
                   Depends: libgl1:amd64 but it is not installable
                   Depends: libglib2.0-0:amd64 but it is not installable
                   Depends: libgtk-3-0:amd64 but it is not installable
                   Depends: libstdc++6:amd64 but it is not installable
                   Depends: libusb-1.0-0:amd64 but it is not installable
                   Depends: gdb:amd64 but it is not installable
E: Unable to correct problems, you have held broken packages.
```

klonyyy, 2024-09-20:
Hey, I suspect this might be due to the wrong architecture (I currently don't build for arm) could you paste here the output of `uname -a` command?

cosmikwolf, 2024-09-20:
ah yea, I am running on arm, its apple sillicon

Linux ubuntu-vm 5.15.0-119-generic #129-Ubuntu SMP Fri Aug 2 20:37:01 UTC 2024 aarch64 aarch64 aarch64 GNU/Linux

klonyyy, 2024-09-20:
I'll put that on my high priority TODO list and add this architecture to the releases soon. I'll let you know when it's ready to test.

klonyyy, 2024-09-20:
Ehh unfortuantely its not as easy as I thought - there are currently no GitHub aarch64 runners available. It would take too long to cross-compile all dependencies and then maintain them by hand so that I can release also aarch64 binary. If you'd like to try we can work out the compilation process together so that people workin on aarch64 can at least build from source by themselves. Gettting this to work should be pretty straightforward - I already compiled the stlink.a library, just need to add some cmake switches.

klonyyy, 2025-01-12:
Closing due to upcoming MacOS support. Please see #102.

closed 2025-01-12 by klonyyy