klonyyy/MCUViewer #82: [Feature Request] UART and simple protocol support

FluxSwitch, 2024-09-21:
### Version/Branch:

1.0.0/ main

### Operating system:

Windows

### Debug probe:

No debug probe

### Details:

Hello,

I am truly impressed by the interactive experience of the waveform display in your design of MCUViewer.

Like you, I work in power electronics domain and I sincerely hope for a software oscilloscope that offers high sampling rates, high refresh rates, and beautiful, intuitive operation. This is particularly crucial when debugging applications like the FOC current loop, where accurately monitoring transient changes is necessary. Your design of MCUViewer perfectly fulfills these needs (once again, I am amazed!).

Currently, MCUViewer can only be used with specific debugging tools. I earnestly hope you will consider adding UART interface support to MCUViewer. This would extend the software's applicability to any MCU, such as Infineon, TI, Renesas, etc. (almost all MCUs come with a UART interface).

Additionally, during power electronics debugging, EMI interference can easily occur. The most commonly used RS232/485 in the industry can effectively solve long-distance signal transmission issues, and this is also based on UART communication.

Finally, if you are interested in adding this feature, please consider the following design suggestions while keeping it as simple as possible:

1. The protocol should include a sequence number or timestamp field to detect packet loss and avoid misinterpretation of waveform data.
2. CRC checksum (for data integrity, this should be essential).
3. Waveforms can be drawn in a ZOH (Zero-Order Hold) step shape, which will facilitate interpreting digital control timing (similar to MATLAB's plotting style).
4. A simple request-response exchange mechanism, akin to the Modbus protocol.
5. Expandable monitoring channel count (typically, 8 to 32 channels is already the limit for oscilloscopes).

This is for your reference. Thank you!

klonyyy, 2024-09-21:
Hello! RS232/485 is something I considered as well. I did not yet implement it as it requires additional work on the targets side, and is not completety non-invasive. Even though I think it's worth to consider it. I'm adding it to the Future TODO list. Cant promise anything time-wise, but if more peope request it it'll definitely happen faster :) 
BTW thanks for the designs suggestions - I will keep these in mind for sure!

Vasili-Sk, 2024-11-27:
i say need some API for any prototcols. i am already thinking about adding my CAN-bus driver to it..

arsenier, 2025-12-24:
I've developed a very simple CAN protocol for communication with BLDC motors for our RoboCup SSL 4-wheeled omni robots. It's called CANFuoco and specification available at:
https://spbunited.github.io/specifications/
Only in russian, however (and with very questionable formatting at the moment, even in PDF).

Nevertheless, I think it has some potential and maybe it could be useful for implementing such a feature.

TL;DR;
Basically, on target device there are a <number> of registers, with different read and write permissions. Message Master->Slave writes to one of the registers, to which Slaves send message Slave->Master, which is a response in specified format.
Format of the response is specified by special "virtual" register MONITOR_SETTINGS, which tells, what registers Master is interested in. At the moment it has very primitive logic, and it contains just one byte, representing register ID to send back. And Slave just sends in response contents of specified register.

Since registers are up to 8 bytes long there are 7 bytes to expand on this functionality (for example one could list up to 8 registers to feed back and slave would for one Master write respond with 8 registers of it's own). Also there are possibility to create "virtual" registers (as per specification for SHAFT_FEEDBACK register, which encapsulates both shaft position and velocity)