klonyyy/MCUViewer #83: MCUViewer: error while loading shared libraries: libjlinkarm.so.7: cannot open shared object file: No such file or directory

margce, 2024-09-23:
### Version/Branch:

1.0.0

### Operating system:

Fedora 40

### Debug probe:

ST-LinkV2

### Details:

I'm running Fedora 40, so I downloaded the rpm package from the release page and install it.
When trying to launch the app I get nothing so decided to investigate launching from the terminal.

Installed package details
```
Installed Packages
Name         : mcuviewer
Version      : 1.0.0
Release      : 1
Architecture : x86_64
Size         : 29 M
Source       : mcuviewer-1.0.0-1.src.rpm
Repository   : @System
From repo    : @commandline
Summary      : MCUViewer built using CMake
License      : unknown
Description  : DESCRIPTION
             : ===========
             :
             : This is an installer created using CPack (https://cmake.org). No additional installation instructions provided.
```

When launching the app I get
```
➜  MCUViewer ./MCUViewer
./MCUViewer: error while loading shared libraries: libjlinkarm.so.7: cannot open shared object file: No such file or directory
```

Even though the jlink library has been installed here

```
➜  MCUViewer ll /usr/local/lib
Permissions Size User Date Modified Name
.rwxr-xr-x@  25M root 17 Sep 05:22  libjlinkarm.so.7
```
 Any reason why the app would not detect the lib it installed?

Thanks

klonyyy, 2024-09-23:
Hello, it seems that Fedora does not use the default /usr/local/lib directory when it looks for installed libraries. Could you try running the attached installer (it should add a *.conf file to /etc/ld.so.conf.d/ and run sudo ldconfig to update cache)?
[MCUViewer-1.0.0-Linux.zip](https://github.com/user-attachments/files/17104875/MCUViewer-1.0.0-Linux.zip)

margce, 2024-09-23:
> Hello, it seems that Fedora does not use the default /usr/local/lib directory when it looks for installed libraries. Could you try running the attached installer (it should add a *.conf file to [...]

hey thanks! for the update, I installed the new rpm and it worked without having to update the cache. Though I haven't try running any debugging session. I'll report if something else fails.

klonyyy, 2024-09-24:
I'll close this thread for now. Please reopen on problems related to installation. Otherwise just open another issue.

closed 2024-09-24 by klonyyy