klonyyy/MCUViewer #84: ELF / AXF File loading fails without message.

r0oland, 2024-09-23:
### Version/Branch:

main

### Operating system:

Windows 10

### Debug probe:

J-Trace PRO V3 Cortex-M

### Details:

I am using MCUXpresso (the NXP SDK) to generate rather large axf/elf files (>4 MB).

When using a smaller example project axf file (1.8 MB), renaming it to elf and loading the variables it works.

When using my custom code base with a large elf file it tries to load the variables for about 1 minute after which no variables are loaded and no error message is generated. 

I can provide the .elf file in a private message if that helps.

r0oland, 2024-09-23:
FYI: The import fails but adding variables manually works, i.e. address are resolved correctly and data can be read and plotted as expected.

klonyyy, 2024-09-23:
Hello! I've worked with larger *.elf files >6Mb and there were no problems - I think it might be related to something else. If you can please provide the file in a private message or my e-mail. Thanks!

r0oland, 2024-09-24:
I just send the elf (and map) files via mail, let me know if I can be of help in finding the problem...

klonyyy, 2024-09-24:
Thanks, it seems this is a GDB bug or something special about yout *.elf file - when running 'info variables' on your elf in GDB's command line it freezes for a few minutes. I'm afraid i cannot do anything about it now. You could try playing with compiler flags, or look for the solution online.

saculu, 2024-09-25:
i have the same problem with the new version. When i load the elf, the application closes.
i can load the elf and work with the previous version.

klonyyy, 2024-09-25:
Could you give more details regarding the versions? You can click 'about' button and copy the commit ID of the working and the faulty one.

saculu, 2024-09-25:
elf not loeading, app crash. --> commit: fc52d03853d5f13bcad32f821db6802d8fde0762
same elf working --> commit: eb3030ff291405aaf3abc42b1ef2cdbd2f49b163

BTW i want to thank you so much . The app is amazing. Traces are working in MCUViewer with Jlink really well

klonyyy, 2024-09-25:
Thanks! Would you be able to share your *.elf file by email? I'd check what's going on there in the afternoon :)

r0oland, 2024-09-25:
TLDR: Change gcc debug option to `-g1`.

@klonyyy Thanks for the info on GDB causing the issue.
I investigated myself a little and could reproduce the issue using `gdb`. Loading the elf worked, but `info variables` or `info functions` would eventually crash with 
`internal-error: virtual memory exhausted`

I updated my toolchain to the latest version (this includes gcc and gdb) but that did not change anything.

I found the solution though, I changed my compiler settings from `-g3` to `-g1` to only produce minimal debug info and now it works! 

I think it makes sense to close this issue for future reference as a solution. It doesn't sound like @saculu issues is related...

klonyyy, 2024-09-25:
That's great, thanks for posting the solution! @saculu I'll close this tread, please make another one for your problem

closed 2024-09-25 by klonyyy