klonyyy/MCUViewer #86: Improve Variable Drag and Drop

r0oland, 2024-09-26:
### Version/Branch:

main

### Operating system:

Windows 10

### Debug probe:

J-Trace PRO V3 Cortex-M

### Details:

It's not possible to drag and drop variables without selecting them first. Might be an issue of the GUI framework you are using, but maybe not?

### Screenshots/Video:

[image]

klonyyy, 2024-09-26:
Yes it is a known issue to me. It is connected to how Imgui handles the selection and then drag and drop feature. I will try to figure it out as I find it annoying as well.

Candas1, 2024-09-26:
This bothers me as well.
But I have the impression that it wasn't behaving like this in early versions.

klonyyy, 2024-09-26:
Yes it wasn't, but you didn't have the select multiple option back then

klonyyy, 2024-09-29:
It should be fixed now. Let me know if it's improved.  https://github.com/klonyyy/MCUViewer/actions/runs/11090383529/job/30812651343

closed 2024-10-01 by klonyyy

r0oland, 2024-10-07:
Thanks for the fix, now working like a charm!