klonyyy/MCUViewer #87: No variables found with .elf import

thencious, 2024-09-29:
### Version/Branch:

1.0.0

### Operating system:

windows 11

### Debug probe:

ST-Link V2

### Details:

No variables found with .elf import
ELF (in ZIP, elf isn't supported)
[firmware.zip](https://github.com/user-attachments/files/17177051/firmware.zip)

### Screenshots/Video:

[image]

r0oland, 2024-09-29:
Have you tried changing the compiler flag for debug symbols to -g1?

That solved it for me.

thencious, 2024-09-29:
Hi, yes i did just now, it's showing a lot of variables now, although not the current sense phase currents for which I'm only seeing these 3 options. I want current.a and current.b from current_sense.getPhaseCurrents(). Any idea what i need to do to get them?
Currently I'm only using:
  // get phase currents
  PhaseCurrent_s current = current_sense.getPhaseCurrents();
  MySerial.print("A: ");
  MySerial.println(current.a);
  MySerial.print("B: ");
  MySerial.println(current.b);
[image]

klonyyy, 2024-09-29:
Please make sure the variable is global i.e. it persists throughout the whole lifetime of the program. Its not possible to visualize variables which are on the stack and get destroyed after the function returns.

closed 2024-10-01 by klonyyy