klonyyy/MCUViewer #91: GUI Improvements

r0oland, 2024-10-10:
### Version/Branch:

main

### Operating system:

Windows 10

### Debug probe:

J-Trace PRO V3 Cortex-M

### Microcontroller:

NXP RT1064

### Details:

I have two suggestions to improve the GUI:

1. Make it possible to move the divider between Variables and Plots (red line in image)
2. Add column to variable view to show last read value. I am aware you have the plot option to show the values in a table but I think it's annoying to not just see the last value...

### Screenshots/Video:

[image]

klonyyy, 2024-10-12:
Hello, the first point is in my opinion totally valid and I'll try to implement it. However I cannot see the purpose of displaying the last value of each variable. The plot view is optimized such that the variables that are not visible are not sampled. Moreover it would be just a duplication of something you can easily do with a table view as you mentioned.

r0oland, 2024-10-14:
I understand your point, but I think my usage of the MCUViewer might be one you have not on your radar...
I am more and more using it as a live debug tool rather than doing printf debugging. So I am not really interested in live plots, as I am mostly looking at the states / transitions / responses. 
Like you said, I can do this with a table, but being able to quickly add many variables and just see the values would make things simpler.

r0oland, 2024-10-15:
I didn't know ctrl+click was working, that makes things easier...It's not a big difference, just slighty more convenient for my usecase. No worries if you don't see it as an improvement...

klonyyy, 2024-10-15:
Alright, let me close the issue for now as it adds too little speedup. Feel free to reopen if needed.

closed 2024-10-15 by klonyyy

r0oland, 2024-10-16:
Don't forget about the first point with the divider though ;-)

klonyyy, 2024-10-16:
Yeah, I still have it on my list, but it might not be that easy to implement. Will mention this issue when it's ready.

r0oland, 2024-10-23:
> Hmm I'm not sure I understand the speedup - in current case the difference is that you have to drag and drop the variables onto the table view? I think it takes only a second to do that, especially with ctrl+click select, and ten you can save the project. Feel free to explain the problem further, maybe I'm not really grasping the idea (you can add some pictures if needed).

I just checked and I am not able to select multiple variables and delete them or drag them anywhere. 
This is on Win10

klonyyy, 2024-10-23:
The delete multiple option is not implemented, however dragging after selecting with ctrl should be working. Could you show me a wideo of it not working? Which branch / version are you on?

r0oland, 2024-10-25:
Apologies, using the keyboard and mouse together does not appear a strength of mine...works as it should. Not sure what I was doing wrong earlier...