klonyyy/MCUViewer #92: Setting what gdb to use

whoopsmith, 2024-10-10:
### Version/Branch:

1.0.1

### Operating system:

Ubuntu 22.04

### Debug probe:

Jlink Plus

### Microcontroller:

Ambiq Apollo4

### Details:

When trying to load variables MCUViewer is trying to use the host 'gdb' but I need it to use either 'gdb-multiarch' or 'arm-none-eabi-gdb'
How do i control what gdb its trying to use?

klonyyy, 2024-10-12:
Hello, 
I understand that there is a problem with the host gdb being not able to parse the elf file you provide? I think the best solution for now is to alias the gdb with one of the ones you've listed. The only solution that comes to my mind is to make the gdb path configurable in the settings. Do you maybe a have better idea?

whoopsmith, 2024-10-13:
> I understand that there is a problem with the host gdb being not able to parse the elf file you provide? 

Correct when my host gdb tries to parse the arm .elf it goes into a tailspin and starts rapidly eating up memory until I either kill it or the OOM kicks in.

Aliasing assumes that I'm not also using the host gdb for other programs which I sometimes need.  Switching the alias or softlink would be tedious although I could perhaps manage that with some scripting and `source` in the settings.  This would likely be a linux only solution as I don't know if that sort of thing is possible on windows.

>  The only solution that comes to my mind is to make the gdb path configurable in the settings. 

I think this would be a fine solution.  You could even have pre-set settings for various platforms given that the 'arm-none-eabi-*' is almost always what the embedded arm compiler tool chain prefix will be.  And then an  'other' setting that allows a custom name in-case there is some really oddball setup.

klonyyy, 2024-10-13:
Sounds like a great idea! Adding to my TODO list. Should be fixed in the next release.

whoopsmith, 2024-10-13:
> Sounds like a great idea! Adding to my TODO list. Should be fixed in the next release.

Amazing.  Thank you.

closed 2024-12-08 by klonyyy