klonyyy/MCUViewer #94: [Feature Request] Variable manipulation

JRAxelrad8, 2024-11-07:
### Version/Branch:

main

### Operating system:

ubuntu

### Debug probe:

ST-Link

### Microcontroller:

STM32H7

### Details:

**My Issue/Question:**
First of all, love the tool!
If possible it would be awesome if it was possible to change a variables value in real time.
For example, if I have a variable named "run test" I'd like to be able to change it from 0 to 1, get the plots and then back to 0.
Thank you in advance.

Candas1, 2024-11-07:
Isn't it what is done here ?
https://github.com/klonyyy/MCUViewer#variable-viewer

klonyyy, 2024-11-10:
I'd say it's already implemented as @Candas1 suggested. Reopen if needed.

closed 2024-11-10 by klonyyy