klonyyy/MCUViewer #96: Texas Instruments - J-TAG Support

oriol98UPC, 2024-11-11:
### Version/Branch:

v1.0.1/main

### Operating system:

Windows 11

### Debug probe:

J-TAG

### Microcontroller:

(Texas Instruments) - C2000 Series

### Details:

**My Issue/Question:**

I recently discovered this project and was immediately impressed. However, I noticed a feature that seems to be missing and could be valuable for many users.

A useful addition would be to allow users to specify and access certain variables directly by name from the .out file within the Debug Server Scripting (DSS) environment. Specifically, I use JTAG with Texas Instruments devices and would benefit from a feature that integrates real-time variable monitoring, similar to MCUViewer with STLINK. This could simplify access without needing exact memory addresses, especially for real-time monitoring setups.

I propose an initial code that I want to test also.

I'd be happy to help or provide more input if needed.

Thanks!

### Screenshots/Video:

Code:

// Define the path to the .out file generated by CCS
var outFile = "C:/path/to/your/file/outfile.out";  // Change this path according to your .out file location

// Create the DSS debug server
var debugServer = ScriptingEnvironment.instance().getServer("DebugServer.1");
var debugSession = debugServer.openSession(".*C28xx.*");

// Connect the device
debugSession.target.connect();

// Load the .out file
debugSession.memory.loadProgram(outFile);

// List of specific variables you want to read by name
var variableNames = ["variable1", "variable2", "variable3"];  // Replace with the names of your variables

// Read and display the values of the specified variables
for (var i = 0; i < variableNames.length; i++) {
    var varName = variableNames[i];
    try {
        var address = debugSession.symbol.getAddress(varName);  // Gets the address of the variable
        var value = debugSession.memory.readData(0, address, 32); // Reads the 32-bit value from page 0
        print(varName + " = " + value);
    } catch (e) {
        print("Error reading variable " + varName + ": " + e);
    }
}

// Disconnect the device and stop the server
debugSession.target.disconnect();
debugServer.stop();

Python Call Script:

import subprocess

# Path to the DSS executable in your CCS installation
dss_executable = "C:/ti/ccs/ccs_base/scripting/bin/dss.bat"  # Ensure this path is correct
script_path = "C:/path/to/your/script/read_variables.js"  # Full path to the JavaScript file

# Run the DSS script
try:
    result = subprocess.run([dss_executable, script_path], capture_output=True, text=True)
    print("DSS Script Output:")
    print(result.stdout)  # Display the output generated by the DSS script
except Exception as e:
    print(f"Error executing DSS: {e}")

klonyyy, 2024-11-11:
Hello, sorry I'm not sure I got the idea right - isn't it possible to parse the elf file for your microcontroller and attach using JLink?

klonyyy, 2024-11-19:
Closing due to inactivity. Please reopen if needed.

closed 2024-11-19 by klonyyy