klonyyy/MCUViewer #98: Can I use a Map file? I don't have access to the ELF.

smarley2, 2024-12-04:
### Version/Branch:

main

### Operating system:

Windows 11

### Debug probe:

ST-Link

### Microcontroller:

STM32G4

### Details:

**My Issue/Question:**

Hi, I only have access to the Map file.
Can I use it instead of the ELF?

klonyyy, 2024-12-04:
Hello, is there a particular reason why only *.map file is available (asking out of curiosity)? I think it's possible eventually, but for now my plan is to develop some other features. We can see how much interest is in this topic from other users.

smarley2, 2024-12-04:
> Hello, is there a particular reason why only *.map file is available (asking out of curiosity)? I think it's possible eventually, but for now my plan is to develop some other features. We can see how much interest is in this topic from other users.

ok, I will talk with the software team and see if they can provide the Elf.
I was looking on the server where the sw is compiled and the files are available.

smarley2, 2024-12-04:
> Hello, is there a particular reason why only *.map file is available (asking out of curiosity)? I think it's possible eventually, but for now my plan is to develop some other features. We can see how much interest is in this topic from other users.

one more question, do I need to ask the sw team to compile it with SWO enable?

klonyyy, 2024-12-04:
For Variable Viewer you only need SWDIO/SWCLK so a minimal SWD connection. If you'd like to use Trace Viewer the SWO pin has to be configured - the trace registers are going to be set up by the MCUViewer.

smarley2, 2024-12-04:
> For Variable Viewer you only need SWDIO/SWCLK so a minimal SWD connection. If you'd like to use Trace Viewer the SWO pin has to be configured - the trace registers are going to be set up by the MCUViewer.

ok, I'm not sure yet, the idea is to debug a control loop that run in 8khz and I want to see some 32bits variables from it in time during operation. I will try with Variable Viewer first to now request change on the sw at the moment.
Thank you.

klonyyy, 2024-12-08:
I'm closing the issue, but please reopen if there is a real need for *.map files support.

closed 2024-12-08 by klonyyy