klonyyy/MCUViewer #101: "Import Variables" shows non-existent array contents at index 1 but none at index 0

amosnier, 2024-12-29:
### Version/Branch:

3650ed7feaff963ca76abcb72423ee220cdbfa69

### Operating system:

Xubuntu 24.04.1 LTS

### Debug probe:

ST-LINK V3

### Microcontroller:

STM32G431RB

### Details:

Maybe there is a simpler way to reproduce this, I'm not sure. But I'm hoping it's still fairly easy with the instructions below.

Prerequisites: have ninja, arm-none-eabi-gcc and a fairly new cmake on your path. At least these are the obvious dependencies that come to my mind.

How to reproduce the issue:

    git clone ssh://git@github.com/amosnier/mcu_viewer_test_case --depth 1
    cd mcu_viewer_test_case
    cmake --preset debug
    cmake --build build-debug/

And then:

- Open build-debug/st_motor_control_01_eclipse.map. Search for FOCVars. In my case it's at address 0x20000518 with size 0x26. It's an array with a single item that is declared in Src/mc_config.c.
- Open MCUViewer. Under "Acquisition Settings", pick the built elf file: build-debug/Src/st_motor_control_01_eclipse.elf.
- Press "Import variables from *.elf". Search for "FOCVars". Only variables under `FOCVars[1]` starting at address 0x2000053e, which is is 0x20000518 + 0x26, are showed. That seems like a bug, since `FOCVars[1]` does not exist (but `FOCVars[0]` does, at address 0x20000518).

Expected behavior: `FOCVars[0]` is showed instead of `FOCVars[1]`.

### Screenshots/Video:

[image]

amosnier, 2024-12-29:
I realize now that the ELF-import function is in beta. I had not realized that, sorry. I have now tried "Add variable" and "Update variable addresses", and that works fine. I have also not attached the actual ELF-file, but I'm hoping that my instructions more than compensate for that.

klonyyy, 2024-12-31:
Hey! Thank you for a solid issue description - yes this is a known issue to me - seems that gdb lists the arrays based on how they're declared, so the number in the brackets is the number of elements. I need to implement a better array handling. For now the best solution is the one that you've found out.

closed 2025-07-19 by klonyyy