klonyyy/MCUViewer #104: Statistics display disappears on acquisition start

dzid26, 2025-01-26:
### Version/Branch:

1.1.0

### Operating system:

Win11

### Debug probe:

-

### Microcontroller:

-

### Details:

**My Issue/Question:**

https://github.com/user-attachments/assets/f145d3d6-8ca8-46c7-9a66-b50d30c726d7

klonyyy, 2025-01-27:
Up until now I thought it is a desired behavior, as the range over which statistics are calculated is gone. Now I think it can stay on top calculating the whole range. Could you describe what behavior you'd anticipate?

dzid26, 2025-01-27:
My expectation at the time was to observe live changes to standard deviation over the whole range (or of what is on the screen). 

At the very least I think the window should be restored when paused.

klonyyy, 2025-05-22:
This should be solved in the 1.2.0 release: [MCUViewer 1.2.0](mcuviewer.com/#downloads)

closed 2025-05-22 by klonyyy