klonyyy/MCUViewer #105: core halt when trace start

TeteDampoule, 2025-01-27:
### Version/Branch:

v1.1.0

### Operating system:

windows 11

### Debug probe:

ST-link V3

### Microcontroller:

stm32g341

### Details:

As soon as I start trace, I have a kind of core halt.
I implemented a sort of watch dog with timers (which are not affected) and it confirms an unexpected delay.
As i debug a motor, it is also easy to hear.

klonyyy, 2025-01-27:
Hello, I myself also noticed these core halts when connecting to a target - this is probably something we cannot change from the probe point of view. Even though I have an idea how to bypass it by turning the acqusition only once and then pausing/starting instead of reconnecting. It requires me to rebuild the UI and the way acquisition is started/stopped. I will for sure take care of it, but most probably in a couple of months.

klonyyy, 2026-08-28:
Resolved in [1.2.9](https://mcuviewer.com/releases/)

closed 2026-08-28 by klonyyy