klonyyy/MCUViewer #106: release application does not start under windows

Maximor19, 2025-02-04:
### Version/Branch:

V1.1.0

### Operating system:

Windows 10 Enterprise

### Debug probe:

ST-link

### Microcontroller:

STM32

### Details:

after installation, Icon is generated on desktop but klicking on it,  GUI does not show up at all..
In the task manager I see the app is started but does not use any resources

klonyyy, 2025-02-04:
Hello, could you try unpacking the executable (*.exe) and try running it from the upacked bin folder? If it's a company PC there might be some restrictions that I'm not yet aware of.

klonyyy, 2025-02-05:
Could you see if you see any logs in C:/Users/[user]/AppData/Roaming/MCUViewer? Note that AppData might be a hidden folder. If there are logs please attach them.

Maximor19, 2025-02-05:
the logfile.txt is empty

Maximor19, 2025-02-06:
that would be great!. Would you be available today ?... Just give me a time....

klonyyy, 2025-02-06:
Could you send me an email (my email is in my GitHub bio) and we can schedule a call, unfortunately I'm quite busy today, and I have to prepare a build for you to test when we're on call ;) I guess tomorrow should be ok.

Maximor19, 2025-02-07:
many thanks for your asap help Peter. Updating openGL solved the issue...
Kind regards, Markus

klonyyy, 2025-02-07:
Glad I could help!