klonyyy/MCUViewer #111: Capture bug

Vasili-Sk, 2025-02-21:
### Version/Branch:

1.1.0

### Operating system:

Windows 10

### Debug probe:

stlink v3

### Microcontroller:

stm32f446

### Details:

Capture sometimes bugs and sets displayed variable to 0
https://youtu.be/ONhdcWS2_38

klonyyy, 2025-02-24:
This can be due to EMI issues affecting the communication - could you make sure the SWD speed is not too high for your setup?

Also please make sure that the variable is not updated in multiple locations in the code - this way you can sometimes sample two different values.

Vasili-Sk, 2025-02-25:
i doubt it is comm issue, because in other scenarios i had one variable also go 0, while others in same graph being okay. 
also restarting logging right away fixes this (temporary) but later it starts to drop to 0. And i checked it from STM viewer, there is no drops in variable inside program.

About multiple locations i am aware of and always using local variables prior saving value to global one.

Vasili-Sk, 2025-02-25:
also in video i show with mouse cursor how it weirdly draws new variables in middle of screen instead of end of it

klonyyy, 2025-02-25:
Oh ok now I see the problem. Is it possible for you to provide a minimal example that I could reproduce? How often does it happen?

Vasili-Sk, 2025-02-26:
Almost every time after changing variables, adding and removing them to/from plot
Here are the settings:
`
[settings]
version = 1
sample_frequency_hz = 2000
max_points = 20000
max_viewport_points = 20000
refresh_on_elf_change = true
stop_acq_on_elf_change = true
probe_type = 0
target_name = 
probe_mode = 0
probe_speed_khz = 10000
probe_sn = 003500273232511939353236
should_log = false
log_directory = 
gdb_command = gdb

[trace_settings]
core_frequency = 160000
trace_prescaler = 10
max_points = 10000
max_viewport_points_percent = 10
trigger_channel = -1
trigger_level = 0.900000
timeout = 2
probe_type = 0
target_name = 
probe_speed_khz = 10000
probe_sn = 
should_log = false
log_directory = `

klonyyy, 2025-06-19:
Hello, please reopen if this issue persists in the new version.

closed 2025-06-19 by klonyyy