klonyyy/MCUViewer #112: [Feature Request] Add API for remote control

mocambo, 2025-02-21:
Hello,
first of all thank you very much for the MCUViewer!
I would like to suggest a new feature I didn't see in other issues: I'm working in the motor control industries like you, but I come from another brand that use its tool to plot variables.
The tool I'm talking about is Freemaster from NXP: 
https://www.nxp.com/docs/en/user-guide/FMSTERUG.pdf
The feature I'm referring is described in cap. 6.2, pag.57/142.
I use this feature for example in my python script to control my MCU, and it save me a lot of time!
I'm asking this because that tool don't work on linux.
What do you think?

klonyyy, 2025-02-24:
Hello, thanks for the proposal. I can see it being implemented in the future, however right now I see it as a low priority feature (let's see if there are more requests from users). Could you describe a specific use case where it's necessary/more convenient? Are we talking of creating a kind of dashboards with this API?

mocambo, 2025-02-25:
Create a dashboard is an idea, but maybe quite complicated.
I was thinking about very small script to automatize test, to check performance, recreate bug: very often it is needed to setup/tune a lot of parameters and so having to do it manually is quite difficult.
I like this feature in FM because I don't need to spend time to manage the elf file and the jtag connection.
Another topic is to control/manage more than a mcu, but this maybe can be done running two or more MCUViewer sessions?
This feature is to manage slow/quite-slow variable, another topic is very-fast variable: here maybe RTT can be a solution.
So too much stuff here, but I agree with you very low priority.

klonyyy, 2025-02-25:
Ok so a simple time-operation list could be the solution? So that you could run this sequence and it would perform all the needed operations? 

I'm currently working on a recorder module (sometimes referenced to as "snapshot" mode) that allows to visualize high frequency data, when SWD speed is limited. More details on this topic soon ;) 

Supporting one than more MCU by running multiple instances is not an official feature, but feel free to test it out ;) I will add it to my TODO list.