klonyyy/MCUViewer #116: I can't see anything but the

ashlynblack, 2025-03-10:
### Version/Branch:

v1.1.0, Branch: main

### Operating system:

Debian Bookworm

### Debug probe:

NA

### Microcontroller:

NA

### Details:

All I can see is the one panel, no tabs. Pressing Ctrl+Tab brings up a list of panels but only the one with "STOPPED trace probe not found" at the top is visible (Trace Viewer?) I think the other panels might be hidden behind it or something? What are the hotkeys to move tabs/panels around? There's no menu options for this?

ashlynblack, 2025-03-10:
[image]

klonyyy, 2025-03-10:
Is imgui.ini located in the same folder as the executable?

ashlynblack, 2025-03-10:
imgui.ini was in my home directory, I deleted it and the view layout went back to its defaults.

klonyyy, 2025-03-10:
Have you installed it using the package or built from sources? Could you show me the structure of the directory with the binary?

ashlynblack, 2025-03-11:
Ah, it was in my home directory as that was my working directory when I launched it from terminal. The issue is resolved but I do think there should be a way to reset the views in-application.

klonyyy, 2025-03-11:
Alright, just to make sure - you've build the application from source or installed it? I'm asking to make sure the installation process is not at fault. The *.ini file should not be installed in your home directory.

ashlynblack, 2025-03-14:
No it's fine, it's because I launched it from terminal with my home dir as the current working directory.