klonyyy/MCUViewer #117: [Feature Request]

ashlynblack, 2025-03-14:
### Version/Branch:

Any

### Operating system:

Any

### Debug probe:

Black Magic Probe

### Microcontroller:

Any

### Details:

Can this be made to work with the Black Magic Probe (or at least an STM32 Black Pill flashed with the Black Magic Debug firmware) in addition to ST-Link and J-Link, please?

klonyyy, 2025-03-15:
It's possible if the probe has a library for the PC side that allows to write and read memory. Currently it has a low priority, but I'll put it on my feature list :)

klonyyy, 2025-06-22:
Since version [1.2.2 beta](https://download.mcuviewer.com/releases/beta/1.2.2_b2f6e24/) the GDB server is supported - this should allow for connection to black magic probe with openOCD. 

For a quick tutorial please follow: https://docs.mcuviewer.com/documentation/Examples/Examples.html#