klonyyy/MCUViewer #118: [Feature Request] Expanding the ways to write variables

DmitriiRPD, 2025-03-17:
### Version/Branch:

V1.10

### Operating system:

Windows 10

### Debug probe:

ST-Link V2

### Microcontroller:

STM32F072, STM32G431, STM32F746

### Details:

Some devices require factory calibration. However, the device does not have a display or buttons. I use an external remote control with a UART interface. But this requires additional actions with the software and hardware.
Another way is to set parameters via the SWD interface using Cube Monitor. But Cube Monitor is an extremely unsuccessful tool.
It would be very nice to add a function to update the value of a variable by moving the slider and clicking on the button. Pressing the button records a predefined value.
This will allow to set the value of a variable in real time, monitor the behavior of the device and write the set value to memory by pressing the button.
With best regard!

### Screenshots/Video:

[image]

klonyyy, 2025-03-19:
Hello, this is already implemented in the new version (coming out soon). Here's how it looks like: 

[image]

does something like this solve your issue? The values to write are kept in the config file so when you load the project the last written values are automatically loaded.

klonyyy, 2025-03-19:
This actually seems too application specific - I think you're looking for a dashboard, not a debug software. For now I don't plan to implement such features, but maybe simple scripting (#112) will be of interest to you (this is a planned feature).

klonyyy, 2025-07-19:
Closing issue due to inactivity - please reopen if needed