klonyyy/MCUViewer #120: GDB needed under Windows 11

baettigp, 2025-04-03:
### Version/Branch:

1.1.0

### Operating system:

Windows 11

### Debug probe:

j-link mini (EDU)

### Microcontroller:

RP2040

### Details:

Hi
I found your program via Elektor Magazine (German edition) thank you very much. I am writing firmware for a raspberry Pi Pico (2040 atm, 2350 soon) under the arduino environment on windows.
I enabled "optimize for debugging" and exported the files. I loaded the elf in MCUViewer and copied the uf2 on the rp2040 in windows explorer. I followed the video for the configuration.
The jlink mini edu is connected to the SWCLK/SWDIO pins . 
However: When I click on run, it shows "STOPPED Could not connect to the target"
I tried both rp2040 0 and 1 variants.

I don't have a gdb installed, (but that was not written in the windows installation instructions?)

Thank you very much for any hints you might have!

ps would it be possible to support the raspberry pi debug probe as well?

klonyyy, 2025-04-03:
Hello, if the variables are found and can be imported using the import window the GDB works fine. Can you confirm using Jflash/any other tool that the problem with connection is on MCUViewers side?

baettigp, 2025-04-03:
Hi @klonyyy 
Thank you very much for your help! After some googling I found the following page: https://kb.segger.com/Raspberry_Pi_Pico
VTref needs to be connected to 3V3 (and for good measure i also connected reset to \bar{run}. Now it is working and I can try and find where my code hangs.
Thank you very much!
Best regards
Pio

closed 2025-04-03 by baettigp