klonyyy/MCUViewer #121: Selected elf path without space

SGanard, 2025-04-09:
### Version/Branch:

main

### Operating system:

Windows 10

### Debug probe:

J-Link

### Microcontroller:

Cortex-M0

### Details:

Is it possible to take into account path with space to avoid tu move elf file?

klonyyy, 2025-04-09:
Hello, yes this is already solved in the new version. Should be released within a month or so.

SGanard, 2025-04-09:
> ***EXTERNAL: This email originated from outside the organization.
>
> Ensure you trust this sender before clicking on any links or attachments.***

Great !
I start to work with your tool, it’s really fantastic! Great job.

Sébastien GOUIN

“This e-mail and any accompanying attachments are confidential. The information is intended solely for the use of the individual to whom it is addressed. Any review, disclosure, copying, distribution, or use of the email by others is strictly prohibited. If you are not the intended recipient, you must not review, disclose, copy, distribute or use this e-mail; please delete it from your system and notify the sender immediately.”

closed 2025-04-09 by SGanard