klonyyy/MCUViewer #124: When using J-Link to connect to the chip, can I use a jlinkscript

Lyubiao, 2025-05-06:
### Version/Branch:

1.1.0

### Operating system:

win10

### Debug probe:

jlink v9

### Microcontroller:

S32K311

### Details:

**My Issue/Question:**

When I'm using this software on the S32K3, the program runs fine when the J-Link isn't connected. However, once I connect the J-Link to the MCU, it enters a hard fault. I encountered a similar issue with the J-Link RTT Viewer before. Can MCUViewer use a J-Link script during the connection process?

### Screenshots/Video:

[image]

klonyyy, 2025-05-17:
Hello, sorry for late response - currently it's not possible to connect with a j-link script, however I might add this option for the next release.

klonyyy, 2025-06-22:
Hello, in [1.2.2 beta](https://download.mcuviewer.com/releases/beta/1.2.2_b2f6e24/) GDB server support was added - you can use it as a temporary solution until JLink script is fully supported.