klonyyy/MCUViewer #126: elf file import, array of structs gets incorrect addresses

ElwinBoots, 2025-05-14:
### Version/Branch:

1.1.0 3650ed7feaff963ca76abcb72423ee220cdbfa69

### Operating system:

Windows 11

### Debug probe:

ST-Link V2

### Microcontroller:

stm32f405

### Details:

[MESC_F405RG.zip](https://github.com/user-attachments/files/20213401/MESC_F405RG.zip)

In the import window I get the variables mtr[1].xxx, but they should be mtr[0].xxx. Note: mtr[1] is not present in the elf file. The addresses I get are for what would be mtr[1], if it would exsist.    

P.S. Overall the MCUViewer is very useful, great work!

### Screenshots/Video:

[image]

klonyyy, 2025-05-15:
Hello, yes the issue is quite known and described also in #101. I have it on my to-do list but I want to handle the arrays better not just make a quick fix on the index :)

ElwinBoots, 2025-05-19:
Thank you for the quick response! Yes, that is indeed exactly the same issue, thanks for the update. I did not see it because  iwas looking for issues with elf in the name.

klonyyy, 2025-06-04:
Closing as #101 duplicate it will be solved for the next release.